mdhiggins/ESP8266-HTTP-IR-Blaster #2: Recalling IR Code issue

Fmkjr, 2017-01-09:
On firmware ESP Basic 3.0.Alpha 65, I have the ESP connected to the wireless at 10.0.0.220. It appears to receive IR codes and save them to locations, for testing I have "" all passwords. when I send 10.0.0.220/msg?code=1 from chrome it responds with "Triggering code 1"  not the IR code which is e0e09966:SAMSUNG:32 it seems to just respond with whatever I put after "code=" but doesn't error check or trigger an actual code or output to the IR.

[default.bas.txt](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/692610/default.bas.txt)

attached is the .bas I am using, for testing I have "" all passwords.
Thanks

mdhiggins, 2017-01-09:
What you're seeing as far as the HTML output is normal behavior.
I just pushed an update that will have the message say "Triggering" and then the IR code even when sending the saved codes (1-6), but the old behavior was just a messaging style choice, it shouldn't have impacted sending the code and does not indicate a problem.

If you're not getting IR output I would double check the wiring. The new update should help with debugged though to ensure the code has saved properly.

mmiscool, 2017-01-09:
The problem you are having is related to the esp basic firmware having not implemented support for sending samsung ir codes.

I noticed this issue and have compiled a new version of the esp basic firmware that has the samsung codes supported. 

The new version is available at esp8266basic.com and should work. Version 3.0a65

Fmkjr, 2017-01-09:
Thank you, new code flashed sends codes. However the IR signal is very weak, I have to be with in 2ish feet for it to work. 
Based on your diagram can I 
1. Add 4 additional IR lads and wire them to in front of each device (example one in front of tv, receiver, etc) 
or
2. Increase the output of the IR led some how. 

I would like to place it about 8ish feet away if I can increase IR output

Thanks for responding so quickly.

mdhiggins, 2017-01-09:
Yes I actually used a 2N2222 transistor and connected my LED to the VIN pin of the ESP8266. This increases the voltage to 5V directly off the USB port and can draw much more current (compared to the tiny 12mA off the GPIO pins)

I'm at work but I'll post a diagram / pic of my latest revision of the board when I can.

mdhiggins, 2017-01-15:
See the V2 readme

closed 2017-01-15 by mdhiggins