mdhiggins/ESP8266-HTTP-IR-Blaster #3: Raw IR

goostoff, 2017-01-10:
Looking through the IRremoteESP8266.h file, we should have the ability to send an int array as raw IR.  How can we accomplish this using the BASIC HTTP-IR-Blaster code?

I'm looking at being able to send Roomba IR clean codes to a 660.  These show up as unknown and it appears to be a series of IR commands, not just a single one.  I've found codes here I'd like to try but they are in a different format than what is used here.

http://www.robotreviews.com/chat/viewtopic.php?t=8682#p64966

mdhiggins, 2017-01-10:
Can’t quite yet

I have a new version coming soon with a custom firmware that supports raw (and encoded roomba commands actually)

Will be up soon

mdhiggins, 2017-01-11:
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/tree/v2

Just pushed an alpha version of the V2 code with Roomba and RAW support and many improvements

Readme still needs works but I put together the gist of getting things set up

goostoff, 2017-01-11:
I'm expecting two more ESP8266's delivered tomorrow.  I will test then.

mdhiggins, 2017-01-15:
V2 now supports RAW IR and Roomba signaling. 

Here are the potential roomba codes:

```
 IR Remote Control
 129 Left 
 130 Forward 
 131 Right 
 132 Spot 
 133 Max 
 134 Small 
 135 Medium 
 136 Large / Clean 
 137 Stop 
 138 Power 
 139 Arc Left 
 140 Arc Right 
 141 Stop  
 
 Scheduling Remote  
 142 Download 
 143 Seek Dock 
 
 Roomba Discovery Driveon Charger 
 240 Reserved 
 248 Red Buoy 
 244 Green Buoy 
 242 Force Field 
 252 Red Buoy and Green Buoy 
 250 Red Buoy and Force Field 
 246 Green Buoy and Force Field 
 254 Red Buoy, Green Buoy and Force 
 
 Roomba 500 Drive-on Charger
 160 Reserved 
 161 Force Field 
 164 Green Buoy 
 165 Green Buoy and Force Field 
 168 Red Buoy 
 169 Red Buoy and Force Field 
 172 Red Buoy and Green Buoy 
 173 Red Buoy, Green Buoy and Force Field
 
 Roomba 500 Virtual Wall
 162 Virtual Wall 
```

closed 2017-01-15 by mdhiggins