mdhiggins/ESP8266-HTTP-IR-Blaster #6: No response from web server from WAN

Jo3yDaPh1sH, 2017-01-15:
My problem is that I cannot access it from outside my network, the ifttt commands get sent by alexa but never make it back to the esp. If I use my External IP or my local ip from my home network it sees it and reports server is running, if I try from my mobile network I get a time out no response. I have the ports forwarded correctly, as the v1 had no issues. I also tried different ports and redoing the forwarding. I checked canyouseeme and the port that is forwarded takes about 5 seconds to respond closed, as any random port reports closed immediately so i think the request is at least making it there. Is there something I am missing or do you have any suggestions?

mdhiggins, 2017-01-15:
That's very strange I can't say I've seen that problem before. If you wanna post a screenshot of your port forwarding settings I can take a look

Jo3yDaPh1sH, 2017-01-15:
[image]
Its very strange, I am going to factory reset my router and re do my settings just to see if there is an changes. Thanks

mdhiggins, 2017-01-15:
Yeah I've got nothing. Your settings all look good not sure why. I suspect it has to do with the router more so than the code or the ESP8266.

Jo3yDaPh1sH, 2017-01-15:
Yea, Its strange as the same router played just fine with the v1 code.

Jo3yDaPh1sH, 2017-01-15:
[image]
Are these the correct settings for ifttt?

mdhiggins, 2017-01-15:
That looks right. And I guess if you try that URL from an external source it doesn't work?

If you take that URL from IFTTT and replace with the local IP does it work? (Locally obviously)

Does just accessing the address without the command bring anything up?

http://pisco.asuscomm.com:6543/

Should bring up a message saying the server is running. 

Does your external IP work? As opposed to the ASUS DDNS

And I would also try disabling the hard coded IP address and let the ESP8266 connect with DHCP then forward to whatever address is assigned and see if that helps. You can do this by commenting out line 38. (WiFi.config...)

Let me know

Jo3yDaPh1sH, 2017-01-15:
Yes it works locally,  I swapped out my router with a router running shibby tomato still no luck.  Locally it works with the lan ip, the wan ip and the ddns name. Reports correctly in serial monitor and does what it is supposed to. It does not work when I try it from my cell phone on the lte network or ifttt. The new ddns is as follows. 

http://joeydaphish.ddns.net:357/

Can you check it from your connection to see if you see server running?

mdhiggins, 2017-01-15:
If you try configuring the ESP8266 to use DHCP and then forward to that IP address does it make any difference?

Jo3yDaPh1sH, 2017-01-15:
seems to have resolved it, hmmm very strange I would much prefer the static ip but I guess this will work!

closed 2017-01-15 by Jo3yDaPh1sH

mdhiggins, 2017-01-15:
Once the dynamic IP is working I would just check what values were auto assigned and then hardcode those values. That's worked for me when I've run into trouble.

Jo3yDaPh1sH, 2017-01-15:
Thanks, I just manually assigned the IP in my router DHCP config, so it shouldn't change, no harm no foul.