mdhiggins/ESP8266-HTTP-IR-Blaster #7: Can't connect to Web Server

chisaipete, 2017-01-15:
I've got the ESP8266 board hooked up to monitor the serial port with correct wifi configuration values.
I upload the sketch and get the following over serial:

`
ü
ESP8266 IR Controller
.WiFi connected
192.168.1.10
HTTP Server started on port 80
Ready to send and receive IR signals
`

But if I now go to my browser and navigate to 192.168.1.10, the browser just sits there at "Connecting..." and then throws an error that the server is taking too long to respond.

I can run other web server example codes from the Arduino IDE with the same configuration and it seems to work just fine, any ideas?

mdhiggins, 2017-01-15:
Anytime I've encountered this problem its usually related to the hardcoding of the IP address. Changing the IP address to something else usually fixes it.

Commenting out line 38 will let the ESP8266 get its IP via DHCP which would be useful for debugging this problem, I would give that a try.

`WiFi.config(ip, dns, gw, subnet); // Enable to hardcode an IP address`

to 

`//WiFi.config(ip, dns, gw, subnet); // Enable to hardcode an IP address`

chisaipete, 2017-01-15:
Ah, could it be bad I set a static IP for my ESP8266?  Could that be messing things up?  Perhaps the server creation code doesn't work well that way?  I could exclusively use DHCP.  I'll check it out later today.

mdhiggins, 2017-01-15:
It's normally fine to set a static IP but occasionally can cause conflicts. Enabling DHCP will help remove that from the equation since you're having issues.

chisaipete, 2017-01-16:
Yeah, it seems to be working now. Hmm, I suppose I can live with that work around, thanks!

closed 2017-01-16 by chisaipete