mdhiggins/ESP8266-HTTP-IR-Blaster #8: RC6 Codes not working.

Jo3yDaPh1sH, 2017-01-19:
Not sure if this is an issue or a request, the program receives and decodes RC6 Codes, but upon sending I get a msg saying code sent, but no blink of the IR. This is the code I am trying which is the power code for an xbox 360 (Grabbed Code from my Harmony)  800F740D:RC6:36   I have tried sending the raw code (the ir blinks) but I get no response from the xbox. 
Thanks

mdhiggins, 2017-01-19:
Pushed an update that should add support for RC5 and RC6, give it a try and let me know

Jo3yDaPh1sH, 2017-01-19:
Do i need any additional libraries to compile this? I get this error in the arduino IDE
 
D:\Downloads\IRController\IRController.ino: In function 'void irblast(String, String, int, int, int, int, int, long int, IRsend)':

IRController:383: error: 'class IRsend' has no member named 'rc5'

           irsend.rc5(data, len);

                  ^

IRController:385: error: 'class IRsend' has no member named 'rc6'

           irsend.rc6(data, len);

                  ^

exit status 1
'class IRsend' has no member named 'rc5'

Jo3yDaPh1sH, 2017-01-19:
I add a send in front of the rc5 and rc6 lines and it compiled, Not sure if it is correct but it fit the format of the other strings so I will give it a go.
 } else if (type == "rc5") {
          irsend.sendRC5(data, len);
      } else if (type == "rc6") {
          irsend.sendRC6(data, len);

mdhiggins, 2017-01-19:
Ah yeah I did that commit too quickly.

Just fixed it, try now

Jo3yDaPh1sH, 2017-01-20:
Seems like the codes are firing off now as the LED blinks, but I don't get a response from the xbox, strange.

mdhiggins, 2017-01-20:
Does your IR blaster work for any other devices?

Does the raw signal work?

Jo3yDaPh1sH, 2017-01-20:
Yes, NEC and raw codes sent to my mistsubisi tv and onyko receiver work
great, I have a json Macro set to turn on and off tv and receiver that I
use daily. Still working on the Xbox and htpc power issue(libreelec).

mdhiggins, 2017-01-21:
Have you tried sending the code to the xbox in the RAW format?

Jo3yDaPh1sH, 2017-01-21:
Yes, no luck. I am going to see if I can find another device that uses the
RC6 codes to see if they are working. I did a little research on the Xbox
360 power code and seems it's not the easiest to get to work correctly.
Thanks for sharing your work and trying to help.

mdhiggins, 2017-01-21:
Alright gonna close this for now since it seems like primarily and xbox related issue. Feel free to update this thread though if you need any further assistance or you find a solution to your problem.

closed 2017-01-21 by mdhiggins

mdhiggins, 2017-01-21:
http://www.righto.com/2010/12/64-bit-rc6-codes-arduino-and-xbox.html

Perhaps this will help. Seems as though you'd need to make some modifications to the RC6 IRremote library since it only goes up to 32 bit and the xbox uses 64 bit codes

arishy, 2017-12-28:
I faced the same issue (RC6) when I was dealing with IRserver from the Library IRremoteesp8266. and I solved it. If you are still trying ...let me know. 

From my experience ( Dec  2017 ) The library can handle 64 bit.