mdhiggins/ESP8266-HTTP-IR-Blaster #10: &out parameter not working when JSON data is passed through a URL

klier, 2017-01-21:
Example URL:

http://192.168.2.75:8081/json?pass=xxx&out=1&plain=[{%22type%22:%22raw%22,%22data%22:[9050,4600,650,550,600,1700,650,1700,650,1700,650,600,600,1700,650,1700,650,650,650,550,650,600,650,550,650,650,650,600,600,1700,650,550,650,600,650,650,600,1700,650,550,650,550,650,1700,650,1700,650,1700,650,1700,650,1650,650,1700,650,550,650,550,600,600,600,650,650,550,600,600,650,550,600,650,600,1700,650,1700,650,1700,650,1700,650,1700,650,1700,650,1700,650,1700,650],%22khz%22:38,%22pulse%22:1}]

It doesn't matter if out is set to 1 or 2, all IR output still comes out of Output #1.

klier, 2017-01-21:
irsend parameters are at default:

IRsend irsend1(4); // Transmitting preset 1
IRsend irsend2(0); // Transmitting preset 2
IRsend irsend3(12); // Transmitting preset 3
IRsend irsend4(13); // Transmitting preset 4

mdhiggins, 2017-01-21:
For the JSON style commands you need to include it in the JSON object itself, not as a URL parameter

klier, 2017-01-21:
Thanks for that pointer.  That's exactly what was wrong...

closed 2017-01-21 by klier