mdhiggins/ESP8266-HTTP-IR-Blaster #9: Sony TV doesn't work

yavinmaster, 2017-01-21:
I have tested my device on a few pieces of equipment and most work.  Two unfortunately don't work, one which comes up as "UNKNOWN" (which I will come back to), but one is recognised as SONY (and it is a SONY TV).

On monitoring the comms port I see:

10:SONY:12
unsigned int  rawData[25] = {2450,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 1250,550, 650,550, 650,550, 650,550, 650};  // SONY 10
unsigned int  data = 10;

10:SONY:12
unsigned int  rawData[25] = {2450,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 1250,550, 650,550, 650,550, 650,550, 650};  // SONY 10
unsigned int  data = 10;

10:SONY:12
unsigned int  rawData[25] = {2450,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 1250,550, 650,550, 650,550, 650,550, 650};  // SONY 10
unsigned int  data = 10;

10:SONY:12
unsigned int  rawData[25] = {2450,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 1250,550, 650,550, 650,550, 650,550, 650};  // SONY 10
unsigned int  data = 10;

10:SONY:12
unsigned int  rawData[25] = {2450,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 650,550, 1250,550, 650,550, 650,550, 650,550, 650};  // SONY 10
unsigned int  data = 10;

I have tried:

http://192.168.1.130:8081/msg?code=10:SONY:12&pulse=5&repeat=1&pass=pass
and
http://192.168.1.130:8081/json?pass=pass&plain=[{%22type%22:%22raw%22,%22data%22:%222450,550,650,550,650,550,650,550,650,550,650,550,650,550,650,550,1250,550,650,550,650,550,650,550,650%22,%22pulse%22:%225%22,%22khz%22:%2232%22,%22repeat%22:1,%22rdelay%22:80}]

and neither works.

My Sony Bluray works fine.

Any suggestions?

mdhiggins, 2017-01-21:
For SONY TVs the pulse signal should be sent 2-3 times with a pdelay of 100ms.

At some point in the code I made 100ms the default pdelay but depending on when you flashed your device that may not be the case.

I would try this:

`http://192.168.1.130:8081/msg?code=10:SONY:12&pulse=3&pdelay=100&pass=pass`

I actually have a SONY TV and am able to send the `10:SONY:12` signal without any issues using that above URL so I don't think there is anything wrong with that code.  

As far as the unknown code goes, some codes may not be recognized and they'll need to be sent as RAW. What device are you sending from?

yavinmaster, 2017-01-21:
Thanks for the quick reply.

I have tried what you have supplied and it still doesn't work (I only
downloaded and installed the code today so it should be up to date).

I actually have two SONY TVs and neither of them like this code.  Any other
thoughts?  The fact that other devices work suggests I have everything
connected correctly.

The UNKNOWN device is an odd one - it is actually a Subwoofer.  The serial
monitor reports:

6793A643:UNKNOWN:32
unsigned int  rawData[19] = {1950,1950, 950,950, 1950,950, 950,950,
950,1950, 950,950, 1950,1950, 950,950, 1950,1950, 950};  // UNKNOWN 6793A643

6793A643:UNKNOWN:32
unsigned int  rawData[19] = {1950,1950, 950,950, 1950,950, 950,950,
950,1950, 950,950, 1950,1950, 950,950, 1950,1900, 950};  // UNKNOWN 6793A643

Thanks.

mdhiggins, 2017-01-21:
What brand is the subwoofer?
And unknown doesn't mean there's an error, it just means that whatever IR protocol the subwoofer remote is using isn't one that the IRremote library has predefined. You should still be able to send commands to the subwoofer using RAW

For the SONY TVs I would try playing with the pdelay value and see if there's a sweet spot that it starts to recognize the code. 

Start with 10ms and go up by increments of 10. The TV I'm testing on is actually a Westinghouse brand that just uses SONY remote codes so perhaps their implementation is a little different. I've read on other sources that 40ms sometimes works. But definitely just start at 10 and go up by 10 and see if you get signal capture.

Also I would like to clarify, do the SONY TVs fail to recognize every code or just the 10:SONY:12 one?

yavinmaster, 2017-01-21:
It's a tsunami subwoofer (quite old now but still sounds good).

I have tried the Sony TV with a pdelay from 10 to 300 in increments of 10
and still no joy unfortunately.  I haven't tried every function on the TV,
but those I have tried didn't work.

mdhiggins, 2017-01-21:
Hm very odd. And what pulse setting are you using?

yavinmaster, 2017-01-21:
I started using a pulse setting of 3, but have even tried 6 and 10.
I have put a scope on the IR receiver and the Sony pulses appear to be  27ms apart.  Setting the output to this doesn't however make it work even though the output from both the IR Blaster and the original remote appear to be similar in appears now.  What I haven't worked out yet is whether I can use the scope to decode the signals to compare those.

yavinmaster, 2017-01-21:
Based on the RAW code I provided for the subwoofer - can you confirm the http code you would use (in case I have the formatting wrong).  This might also be the route I have to take for the Sony TV if I have it wrong currently.

yavinmaster, 2017-01-21:
I can't currently decode the IR codes directly, but the attached images show a different patter from the original remote and the "same" code being sent by the IR blaster.
The original has a long header followed by 12 pulses, the 8th one being "long".
The IR Blasters has a long header followed by 11 pulses, the 7th one being "long".

[image]
[image]

mdhiggins, 2017-01-21:
Can you post both the serial raw output and your JSON object you're passing for that signal?

yavinmaster, 2017-01-22:
The the subwoofer;
serial monitor sees:
6793A643:UNKNOWN:32
unsigned int  rawData[19] = {1800,1850, 900,950, 1800,900, 900,950, 900,1850, 900,950, 1800,1800, 900,900, 1800,1850, 900};  // UNKNOWN 6793A643

6793A643:UNKNOWN:32
unsigned int  rawData[19] = {1800,1850, 900,900, 1800,950, 900,950, 900,1850, 900,950, 1800,1850, 900,950, 1800,4200, 550};  // UNKNOWN 6793A643

Are there mean't to be spaces characters between some of the numbers?

Command tried (and didn't work):
http://192.168.1.130:8081/json?pass=pass&plain=[{%22type%22:%22raw%22,%22data%22:%221800,1850,%20900,950,%201800,900,%20900,950,%20900,1850,%20900,950,%201800,1800,%20900,900,%201800,1850,%20900%22,%22khz%22:%2232%22,%22repeat%22:1,%22rdelay%22:80},{%22type%22:%22raw%22,%22data%22:%221800,1850,%20900,900,%201800,950,%20900,950,%20900,1850,%20900,950,%201800,1850,%20900,950,%201800,4200,%20550%22,%22khz%22:%2232%22,%22repeat%22:1,%22rdelay%22:80}]

yavinmaster, 2017-01-22:
OK - back to the SONY.  I have loaded the IR decoding program from https://learn.adafruit.com/using-an-infrared-library/hardware-needed onto another ardunio. 

When I use the original remote I get:

Decoded Sony(2): Value:10 (12 bits)
Raw samples(26): Gap:13946
  Head: m2250  s700
0:m500 s650	1:m500 s650		 2:m500 s700	3:m450 s700		 
4:m500 s650	5:m500 s650		 6:m500 s700	7:m1050 s700		 
8:m500 s650	9:m500 s650		 10:m500 s650	11:m500
Extent=16800
Mark  min:450	 max:1050
Space min:650	 max:700

If I then transmit the code from the IR Blaster, I get:

Decoded Unknown(0): Value:0 (0 bits)
Raw samples(24): Gap:40550
  Head: m2950  s650
0:m550 s650	1:m600 s600		 2:m600 s650	3:m550 s650		 
4:m550 s650	5:m550 s650		 6:m1150 s650	7:m550 s650		 
8:m550 s650	9:m600 s600		 10:m600
Extent=16850
Mark  min:550	 max:1150
Space min:600	 max:650

mdhiggins, 2017-01-22:
I think at this point we're getting beyond my understanding. The program uses the IRremote Arduino library to handle most of this stuff. I am unable to replicate your problem as my 10:SONY:12 codes are working as are my RAW codes as least on the devices that I have for testing. 

I would make a post on that Github page and see if he has any insight into what's going on.  This problem seems to go deeper than just the code I've written on top of the IRremote library.

mdhiggins, 2017-01-22:
If I take two ESP8266 boards and having one sending the 10:SONY:12 code the other one seems to receive it correctly and displays the 10:SONY:12 as the received code.

This is using my blueprint however, not an outside library so everything is still based on the IRremote library.

yavinmaster, 2017-01-22:
Thanks - I'll have to find some time to dig deeper.  Can you confirm I have
formatted the raw output correctly?

mdhiggins, 2017-01-22:
Do you have the URL before encoding the special characters? It's hard to tell exactly after encoding 

Sent from my iPhone

> On Jan 22, 2017, at 13:34, yavinmaster <notifications@github.com> wrote:
> 
> as

yavinmaster, 2017-01-22:
http://192.168.1.130:8081/json?pass=pass&plain=[{"type":"raw","data":"1800,1850,
900,950, 1800,900, 900,950, 900,1850, 900,950, 1800,1800, 900,900,
1800,1850,
900","khz":"32","repeat":1,"rdelay":80},{"type":"raw","data":"1800,1850,
900,900, 1800,950, 900,950, 900,1850, 900,950, 1800,1850, 900,950,
1800,4200, 550","khz":"32","repeat":1,"rdelay":80}]

mdhiggins, 2017-01-23:
I actually do see a little problem, the raw values should be an array not wrapped in quotes.
Change to this:
```
http://192.168.1.130:8081/json?pass=pass&plain=[{"type":"raw","data":[1800,1850,
900,950, 1800,900, 900,950, 900,1850, 900,950, 1800,1800, 900,900,
1800,1850,
900],"khz":"32","repeat":1,"rdelay":80},{"type":"raw","data":[1800,1850,
900,900, 1800,950, 900,950, 900,1850, 900,950, 1800,1850, 900,950,
1800,4200, 550],"khz":"32","repeat":1,"rdelay":80}]
```

Also as a rule of thumb I think almost all pretty much all IR signals use 38khz as their operating frequency, not sure if you're intentionally using 32 but that may be part of the problem.

yavinmaster, 2017-01-23:
Bingo!!!

The 32/38 was a typo on the sample I sent you (I did previously have this
correct I am sure).  My mistake was the array being in quotes not an array.

Using this method, the "strange" subwoofer codes work and I can also use
the raw codes to get the Sony TV to work too.  Strange that the "proper"
Sony codes don't result in the correct output, but so long as I have a
method to make it work.

Thanks for your help.

mdhiggins, 2017-01-23:
Excellent! Glad we got something working for you. Very strange that the normal code isn't working but perhaps it's just a strange anomaly.

closed 2017-01-23 by mdhiggins