mdhiggins/ESP8266-HTTP-IR-Blaster #12: Resistors

Flinst0n, 2017-01-25:
Dear Michael Higginis,

can you give rough estimates for the resistors you've used?

Thank you very much for your project!

Cheers
Alex

mdhiggins, 2017-01-25:
The resistor values vary quite a bit depending on what LED you're using.

For the resistor going to the transistor I used a 1k ohm transistor. 

For the LED there are equations you can do to calculate your resistance. The LED aim using can go up to 1A of current for short bursts. 

I have mine connected to the 5V USB power source and the voltage drop is 1.4V for the LED I used a 15 ohm resistor for a 250 mA current across the LED which gives excellent range. 

Ledcalc.com is a useful tool.

Flinst0n, 2017-01-25:
Thank you! :) I slowly start to remember my Electronics 1 course...

My LED has forward voltage U_F = 1.35V (typical) and forward current I_F=100mA. So at a supply voltage of 5V I need a 39Ohms (thanks to ledcalc.om)

closed 2017-01-25 by Flinst0n