mdhiggins/ESP8266-HTTP-IR-Blaster #13: crash loop from trigger

Fmkjr, 2017-01-26:
On the current code with these includes
#include <IRDaikinESP.h>
#include <IRremoteESP8266.h>
#include <IRremoteInt.h>
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
 I get a crash loop after 4 -10 triggers
 a serial dump with comments are as follows
[Serial Dump from Crash.txt](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/731275/Serial.Dump.from.Crash.txt)
A power cycle is the only way to restart the device.

Thanks M

mdhiggins, 2017-01-26:
Looks like you're including extra things that don't need to be there, not sure why you've added those

Fmkjr, 2017-01-26:
What should I remove.
M

mdhiggins, 2017-01-26:
I would only include the things that are on the blueprint on github, no need for extras
```
#include <IRremoteESP8266.h>
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
```

Did you manually add those other imports for a reason? And were there any other changes you made to the blueprint aside from the wifi settings/ports/passes

Fmkjr, 2017-01-26:
No changes to the blueprint other than the includes Ill remove and re test

Fmkjr, 2017-01-26:
Removed the includes. It is looping on boot now. I think there might be a power issue with the esp based on this part of the serial dump 

   ¦ESP8266 IR Controller

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset

mdhiggins, 2017-01-26:
I would double check your physical wiring and make sure its all appropriate

I’ve occasionally run into issues with ESP being pushed too far into the breadboard and having intermittent shorts and reboots/boot loops, pulling it out of the breadboard partially has fixed this. Additionally try booting the ESP by itself without any of the connections plugged in (isolated, not in the breadboard) and see if you can flash it. 

I suspect this is a hardware related issue.

Fmkjr, 2017-01-27:
I pulled the ESP out of the board and re-installed it with 5% of the pins showing and it has been running for a hour. It seems you where right on the money with that

closed 2017-01-27 by Fmkjr