mdhiggins/ESP8266-HTTP-IR-Blaster #14: Westinghouse TV Showing up SONY and doesn't work

danodemano, 2017-01-30:
Hello,

After my previous post the cats managed to break my Visio TV.  :(  Anyway I've replaced it with a Westinghouse but cannot get the power to work.  The code comes across like:

```
A90:SONY:12
unsigned int  rawData[25] = {2450,600, 1250,600, 700,600, 1250,600, 700,600, 1250,600, 700,600, 700,600, 1250,600, 700,600, 700,600, 700,600, 700};  // SONY A90
unsigned int  data = A90;
```

However trying to use A90:SONY:12 with all combinations of pulse/pdelay doesn't seem to work.  The one thing I DO notice with the Westinghouse remote is that when using the camera on my phone the IR pulse looks like a constant stream of light.  Usually I can see the pulses but with this remote I cannot.

Any thoughts on how to get this to work or ideas for next troubleshooting steps?

mdhiggins, 2017-01-31:
I use a Westinghouse with the same Sony codes. Pulse of 2 with a pdelay of 100ms works for my TV 100% of the time.

Fmkjr, 2017-01-31:
I have a Cisco made Fios tv box and have found that if I send this { type : raw , data :[9100,2250, 550], khz :38, pulse :1}, before it helps (it appears to be a dummy code with no purpose). I send channels like "689 enter"  and without the dummy in-front the Cisco box just reads 89 or sometimes 9 but never 689.

danodemano, 2017-01-31:
@mdhiggins and @Fmkjr thank you very much for the assistance!  However neither suggestion worked.  It still refuses to toggle the TV.  I tried it from just a few feet away just to ensure that it wasn't a limit with the distance.  Any other thoughts.

mdhiggins, 2017-02-01:
Does it work with other devices? Can you see the LED lighting up with a camera?

danodemano, 2017-02-01:
Yes, it worked with my previous Visio and works currently with my Onkyo amp.  And yep I see the LED light with my phone camera.

mdhiggins, 2017-02-01:
Hm I'm not sure what to advise at this point. I know its possible with Westinghouse TVs since mine works but I imagine there could be some variability between models. And you said you've tried messing with the pulse and pdelay options without success

You could just setting a pdelay of 10ms and a pulse of something really high like 15-20 and seeing if that works then narrowing from those settings.

Another thing to consider is the wavelength of the remote. Most TVs use 940nm but not all do, there can be some variability. You can probably use the FCC ID of the remote to see what wavelength it transmits on

mdhiggins, 2017-02-01:
As far as your RAW code not working, if you could post an unencoded URL it'd be much easier for me to debug and see if there was a problem there

danodemano, 2017-02-02:
@mdhiggins I tried a whole bunch of different pdelays with pulses.  I even tried a pdelay of 0 with like 100 pulses and I still couldn't get it to do anything.  

I was going to try and look up the FCC ID to confirm but can't find it anywhere in the manual or on the remote.  It might be on the TV but hard to tell since it's against the wall.

As for the raw code, this is it:

`http://192.168.9.201:8081/json?pass=pass&plain=[{"type":"raw","data":"[2450,600, 1250,600, 700,600, 1250,600, 700,600, 1250,600, 700,600, 700,600, 1250,600, 700,600, 700,600, 700,600, 700]","khz":38,"pulse":10}]`

mdhiggins, 2017-02-02:
Don't wrap the array of codes in quotes like you're doing, that's probably causing it to fail
```
http://192.168.9.201:8081/json?pass=pass&plain=[{"type":"raw","data":[2450,600, 1250,600, 700,600, 1250,600, 700,600, 1250,600, 700,600, 700,600, 1250,600, 700,600, 700,600, 700,600, 700],"khz":38,"pulse":10}]
```

closed 2017-02-17 by mdhiggins

danodemano, 2017-02-19:
hey @mdhiggins - I know you closed this out (which is fine, I hadn't gotten back to it) but wanted to post my ultimate solution just in the event someone else finds this.  I ended up having to monkey with the khz value.  Seems like a 50 nails it every time.  So this was my final URL:

`http://192.168.9.201:8081/json?pass=pass&plain=[{"type":"raw","data":[2450,600, 1250,600, 700,600, 1250,600, 700,600, 1250,600, 700,600, 700,600, 1250,600, 700,600, 700,600, 700,600, 700],"khz":50,"pulse":10}]`

mdhiggins, 2017-02-19:
Thanks for sharing that info, that’s helpful for future users that run into this issue

Wonder if this is a thing the newer Westinghouse TVs are doing