mdhiggins/ESP8266-HTTP-IR-Blaster #16: Keep getting "applet skipped" when I try to run the trigger

cjraymond, 2017-02-10:
Hey,
So I am pretty new to these microprocessors and programs, but I decided to take on the project. Everything's working perfectly until I try to send the trigger from my echo dot. Once I send it nothing happens, and if I view the Applet activity is says "Applet skipped" every time I try to run it. I've gone to my ip and the http server is up.

I noticed that there are options I get in the maker channel that are not shown in your screenshot of it. What should I set my method and content type to? Is there anything I should put in the body?

Thanks so much for posting this project and your code!

mdhiggins, 2017-02-11:
Can you post your IFTTT settings?

cjraymond, 2017-02-11:
This is a basic one for my vizio TV. I have one for my Roku too. It seems that when I use my external or internal IP neither changes anything.
[image]
There is nothing in the body section.

mdhiggins, 2017-02-11:
If you manually enter that URL in the browser does it work correctly?

If yes, does that URL work in a browser from a device outside your local network (like on your phone when not on WiFi?)

cjraymond, 2017-02-11:
I thought I had it working before, but now when I try to reach the external IP from my browser (on WiFi) I can't reach it, and same for phone on data. I could probably figure that out within a few minutes, or if you have any suggestions that would be great.

mdhiggins, 2017-02-11:
Sounds like the port isn’t be forwarded properly for the external IP address. I would check your router settings.

You can also try to change the local IP address on the blueprint and adjust the router settings. I’ve had that randomly be an issue that can be resolved with this change.

cjraymond, 2017-02-11:
So, I changed the IP as well as the port to try to see if I could get that to work. I got to the page from a computer on my WiFi and my external IP, but not on my phone on data. I read from another issue here that they weren't able to get it to work from their phone data? I'm not sure.

cjraymond, 2017-02-11:
Okay! So I went back and did some stuff that you had told others to do in the past closed issues and it worked. I can access it from outside the network and the Applet is running. Thank you so much for you time, and helping me. I really appreciate it. This project is awesome, and it's really great how you are responding so quickly to everyone. Again, thanks so much!

closed 2017-02-11 by cjraymond