mdhiggins/ESP8266-HTTP-IR-Blaster #19: Roomba IR

Johnwulp, 2017-03-02:
Hi, i am using this sketch to get my Roomba start cleaning. In the wiki the code 136 should be the command to start cleaning. How do i call my ESP01 to send the right command? I tried `http://192.168.2.8/msg?pass=go&type=roomba&data=136` without succes.

mdhiggins, 2017-03-03:
Try

http://192.168.2.8/msg?code=136:ROOMBA&pass=go&pulse=3

Johnwulp, 2017-03-03:
Thanks, no success either. Going to grab another ir led and test again tonight

Johnwulp, 2017-03-03:
Well, left, right and forward action is working from my harmony one universal remote. Also the clean action doesnt work from here. Guess my roomba is somehow different. Using model 875. Thanks anyway!

closed 2017-03-03 by Johnwulp

ronnicek, 2017-08-03:
Hi, did you get it working somehow? :)

Johnwulp, 2017-08-03:
No, never got it working. Just pressing the clean button manually :-)

ronnicek, 2017-08-03:
I will let you know if I will get it working.. because pressing button is so 90'ish :D

ronnicek, 2017-08-03:
So following code is working with Roomba https://gist.github.com/ronnicek/28c2b004ee5f469ce264b1702ad19cc1 (I edit line 138 - but you can make diff from https://gist.github.com/madc/e6a06abaef28983d66bc to see what I changed :)

I have wemos d1 mini pro, with this LED (https://www.aliexpress.com/item/1pcs-Infrared-Emission-Module-IR-Transmitter-For-Arduino-Electronic-Blocks-new-Free-shipping/32655173182.html) but I have to have LED really, really close (have that on Roomba for testing). So I need to do some another way how to make LED more powerfull (I bought 10pcs 5mm IR Infrared LED 940nm Lamp Transmitting Tube Emitting Diode) so there is space for improvement :)

Try it and let me know :)

Johnwulp, 2017-08-03:
Thanks for your contribution! I have recycled my wemos d1. I will order another one, and some bright IR LEDs, and will try again.

ronnicek, 2017-08-09:
Hi,

so I had some time to play with that more.. and I find out, that Roomba after some time goes to something like "deep-sleep" mode.. because it do not react on power on signal.. I will try to figure out and let you know when I will have something :)

mdhiggins, 2017-08-09:
I have an old 500 series Roomba and have no issues. Can issues commands after days of inactivity without problems. Granted the only command I ever send is for it to start

My command looks like this, I did need pulse 3 to make it reliably turn on 100% of the time
`msg?code=136:ROOMBA&pulse=3`

ronnicek, 2017-08-09:
It's true that I didnt check this code, I just found this issue on search engine :) Can you share what electronics are you using (diode etc. ?).

I will try to upload this to Wemos and lets see if it will work :)

ronnicek, 2017-08-10:
So I found out this - http://imgur.com/a/1Ij7V :)

Johnwulp, 2017-08-18:
Well, I have a few fresh wemos d1 mini's And the good news is, that it is working. The problem on my side was that the code to start cleaning is 134 instead of 136.

ronnicek, 2017-08-19:
Great to hear! From how distance it works?

Johnwulp, 2017-08-19:
Well, I’m sending commands from 1 meter right now, with 1 5mm ir led. I don’t use a transistor, but I connected the led directly to the gpio port. I bought 50 leds for 2 dollar, so I have some spares 👍

shnax, 2017-08-19:
@ronnicek Did you figure out a way around that "deep sleep" issue? I'm running into the same thing with a roomba 652.

ronnicek, 2017-08-19:
@shnax nope.. but when Roomba is in dock, then you can start it with single signal (see http://imgur.com/a/1Ij7V)

raptordemon, 2017-08-29:
i can't for the life of me make this work, i have tried:

/msg?code=136:ROOMBA&pulse=3
/msg?code=136:ROOMBA&pulse=3&repeat=3&rdelay=250
/msg?code=134:ROOMBA&pulse=3
/msg?code=134:ROOMBA&pulse=3&repeat=3&rdelay=250
/msg?code=135:ROOMBA&pulse=3
/msg?code=135:ROOMBA&pulse=3&repeat=3&rdelay=250
/json?plain=[{'data':[2850,950, 950,2800, 950,2800, 950,2800, 2800,950, 950,2800, 950,2800, 950], 'type':'raw', 'khz':38, 'repeat' : 5}]

the last code was learned from my logitech harmony remote that was able to control the rooomba.

the only thing i can think of at this point is my IR led isn't the correct type for the roomba. it works with the TV

any ideas?

Johnwulp, 2017-08-29:
In using:

http://blaster.local/msg?code=136:ROOMBA&pass=GO&pulse=3

I have a password set, you don’t I guess from your url?

I have these ir leds. http://s.aliexpress.com/FfeMZZFz

raptordemon, 2017-10-04:
2 questions for you guys, 
1) where are you aiming the IR led to get the roomba?

2) can someone share a link to some leds that are known working with the roomba?
thanks 
Raptor

mdhiggins, 2017-10-05:
I'm using this one:
https://www.amazon.com/gp/product/B00ULB0U44/

And I point it from above right at the little circle at the front of the Roomba that's slightly elevated

raptordemon, 2017-10-08:
Hmm still doesn't work, moving on to checking the resistor values

mdhiggins, 2017-10-08:
What model roomba?

raptordemon, 2017-10-08:
its a 560 roomba, new resitors didnt work either

mdhiggins, 2017-10-08:
I’m using a 535 not sure if the remote codes are compatible or not. Far as I know they should be

raptordemon, 2017-10-09:
i expect that they are, it works when i use my logitech harmony 880 to control it, i even tried learning the codes and that didn't work either.... the esp8266 can control my TV.

honestly im at a loss

mdhiggins, 2017-10-09:
I was reviewing my Roomba power command and it looks like I had added `&pulse=3` and forgot to mention that. Now that I noticed I'm remembering that it wouldn't run without that pulse. I would try that and see if it helps, if 3 doesn't work just gradually increase and see (wouldn't go higher than 10)

mdhiggins, 2017-10-09:
The comments on this article have some other suggestions:
https://astrobeano.blogspot.com/2013/10/roomba-620-infrared-signals.html

Some people were changing the timings around from 3/1 to 3/1.1, others found that code 149 worked on some other models

Another thought is that its possible your docking station (which also uses an IR LED) is too bright and overwhelming the signal of your current LED, I would try sending the signal when its not near the docking station or with the docking station for testing purposes. That would make the most sense and would explain why the Harmony (which is a brighter device) still works. If that's the case then pushing the current or getting a couple IR LEDs to boost the signal might be worthwhile

The LEDs that I linked earlier that go up to 1A of current are a good option for really pushing the brightness

raptordemon, 2017-10-10:
can you elaborate on this: "Some people were changing the timings around from 3/1 to 3/1.1"
i tired the new codes and changing the pluses to 10 no luck :(

mdhiggins, 2017-10-10:
Did you try moving the Roomba away from its docking station or unplugging the docking station temporarily to see if helped?

And the timing codes are here: (line 1173/74)
```
  unsigned int one_pulse = 3000;
  unsigned int one_break = 1000;
```
People in the comments were changing the break to 1100 etc

joostwestra, 2019-04-12:
http://youserver.yourport/json?pass=yourpass&plain=[ { "type":"raw", "data":[2831, 886, 972, 2709, 944, 2711, 943, 2710, 2743, 893, 958, 2723, 931, 2722, 927, 19304, 2811, 897, 954, 2726, 927, 2726, 927, 2726, 2747, 889, 966, 2714, 942, 2710, 941], "khz":38, "pulse":3 } ]

While al the programmed Rooms commands with multiple repeats did not really work. Maybe the 19304 sleep time is critical.