mdhiggins/ESP8266-HTTP-IR-Blaster #21: Problem with receiving codes

mihalski, 2017-03-22:
I'm trying to use a Vishay TSOP34838 (http://www.vishay.com/docs/82489/tsop322.pdf) infrared receiver (recommended by AnalysIR website) but it seems to be picking up a LOT of random IR codes when powered at 5v (as well as legit, repeatable ones that I generate) and nothing at all when powered by 3.3V.

I've tried on both a Wemos D1 mini and an Adafruit HUZZAH Feather with the same results.

Do you have any suggestions as to why this might be? Or what I need to do to fix it? I'm using GPIO5.

mdhiggins, 2017-03-22:
I've had occasional noise when receiving a code where some random garbage would appear transiently but the real code is usually evident after scanning a few times. Reflective surfaces and lots of ambient light sources seem to increase the rate of this but its still pretty minimal

Are you still ultimately able to determine the code you're trying to capture?

And mine usually works fine off of 3.3V

mihalski, 2017-03-22:
Here's what about 30 seconds looks like in a dark room with NO IR signals being sent (to my knowledge):
```
Controller
.WiFi connected
192.168.1.99
HTTP Server started on port 8081
Ready to send and receive IR signals
FEAC02E5:UNKNOWN:32
unsigned int  rawData[5] = {100,11100, 200,750, 100};  // UNKNOWN FEAC02E5

B994247F:UNKNOWN:32
unsigned int  rawData[11] = {150,450, 100,4400, 100,1000, 100,2600, 150,6450, 100};  // UNKNOWN B994247F

4CB0FADF:UNKNOWN:32
unsigned int  rawData[5] = {150,200, 100,750, 150};  // UNKNOWN 4CB0FADF

FCABFFBF:UNKNOWN:32
unsigned int  rawData[5] = {100,250, 150,1950, 100};  // UNKNOWN FCABFFBF

68AE654:UNKNOWN:32
unsigned int  rawData[9] = {150,200, 150,700, 100,300, 150,950, 100};  // UNKNOWN 68AE654

22AE7A29:UNKNOWN:32
unsigned int  rawData[5] = {150,200, 150,7350, 150};  // UNKNOWN 22AE7A29

8BDFACCB:UNKNOWN:32
unsigned int  rawData[13] = {150,6650, 150,1100, 150,250, 100,950, 100,2000, 100,7150, 150};  // UNKNOWN 8BDFACCB

24AE7D4F:UNKNOWN:32
unsigned int  rawData[5] = {150,5150, 150,950, 150};  // UNKNOWN 24AE7D4F

22AE7A29:UNKNOWN:32
unsigned int  rawData[5] = {150,250, 150,4700, 150};  // UNKNOWN 22AE7A29

FEAC02E5:UNKNOWN:32
unsigned int  rawData[5] = {100,2800, 150,1750, 100};  // UNKNOWN FEAC02E5

5DDE745C:UNKNOWN:32
unsigned int  rawData[9] = {150,950, 150,9500, 200,850, 100,950, 100};  // UNKNOWN 5DDE745C

4CB0FADF:UNKNOWN:32
unsigned int  rawData[5] = {150,1000, 100,3800, 150};  // UNKNOWN 4CB0FADF

71FE45B0:UNKNOWN:32
unsigned int  rawData[9] = {150,12450, 150,1050, 150,250, 150,7150, 100};  // UNKNOWN 71FE45B0

23746BE5:UNKNOWN:32
unsigned int  rawData[15] = {150,950, 150,200, 150,2850, 100,2850, 100,1900, 150,3550, 150,7300, 150};  // UNKNOWN 23746BE5

257F024:UNKNOWN:32
unsigned int  rawData[21] = {150,6950, 100,1750, 150,1000, 100,2250, 100,5100, 100,950, 200,900, 100,950, 150,850, 150,950, 100};  // UNKNOWN 257F024

1410D334:UNKNOWN:32
unsigned int  rawData[7] = {200,1000, 150,250, 100,2000, 100};  // UNKNOWN 1410D334

71FE45B1:UNKNOWN:32
unsigned int  rawData[9] = {150,7350, 150,1000, 150,200, 150,9900, 150};  // UNKNOWN 71FE45B1

4CB0FADD:UNKNOWN:32
unsigned int  rawData[5] = {200,200, 150,2850, 100};  // UNKNOWN 4CB0FADD

24AE7D4F:UNKNOWN:32
unsigned int  rawData[5] = {150,1000, 150,250, 150};  // UNKNOWN 24AE7D4F

4CB0FADF:UNKNOWN:32
unsigned int  rawData[5] = {150,200, 100,1250, 150};  // UNKNOWN 4CB0FADF

22AE7A28:UNKNOWN:32
unsigned int  rawData[5] = {150,1350, 150,2950, 100};  // UNKNOWN 22AE7A28

6BDBBF60:UNKNOWN:32
unsigned int  rawData[7] = {150,250, 100,1000, 100,8850, 150};  // UNKNOWN 6BDBBF60

5DD7C30A:UNKNOWN:32
unsigned int  rawData[15] = {150,200, 150,900, 150,900, 100,1000, 100,3950, 100,2950, 100,13550, 150};  // UNKNOWN 5DD7C30A

9C07722A:UNKNOWN:32
unsigned int  rawData[13] = {150,1000, 150,200, 100,950, 100,950, 150,1900, 150,900, 150};  // UNKNOWN 9C07722A

46172D8F:UNKNOWN:32
unsigned int  rawData[7] = {150,250, 150,1000, 100,6050, 100};  // UNKNOWN 46172D8F

E9E70E8C:UNKNOWN:32
unsigned int  rawData[7] = {150,8350, 150,200, 150,4650, 150};  // UNKNOWN E9E70E8C

22AE7A28:UNKNOWN:32
unsigned int  rawData[5] = {150,3700, 150,7100, 100};  // UNKNOWN 22AE7A28

6F788F20:UNKNOWN:32
unsigned int  rawData[23] = {150,1100, 150,200, 150,2000, 100,950, 150,900, 150,900, 150,900, 100,950, 100,950, 100,950, 100,950, 100};  // UNKNOWN 6F788F20

22AE7A29:UNKNOWN:32
unsigned int  rawData[5] = {150,1000, 150,6850, 150};  // UNKNOWN 22AE7A29

81AC2DA6:UNKNOWN:32
unsigned int  rawData[9] = {100,14100, 150,200, 100,1250, 150,14500, 150};  // UNKNOWN 81AC2DA6

6ADBBDCE:UNKNOWN:32
unsigned int  rawData[7] = {150,1300, 100,4950, 100,4000, 100};  // UNKNOWN 6ADBBDCE

84071B38:UNKNOWN:32
unsigned int  rawData[11] = {200,150, 100,1100, 300,300, 150,900, 150,1150, 200};  // UNKNOWN 84071B38

22AE7A28:UNKNOWN:32
unsigned int  rawData[5] = {150,1200, 150,2850, 100};  // UNKNOWN 22AE7A28
```
And I get NOTHING when running at 3.3V, so I have to run at 5V.

I was trying to capture Mitsubishi A/C code and do to it's size and complexity I couldn't pick it out of the noise. I'd love to figure out how to get a clean signal AND have it work at 3.3V as it clearly should.

Will try running off battery and see what happens in other rooms when I get the chance tomorrow.

consolacion, 2017-04-11:
Just wondering if you were able to solve your problem/issue

mihalski, 2017-04-11:
Not as of yet.. I do have one suspicion which I plan to investigate tomorrow, and I'm waiting for a bunch of modules from AnalysIR so that I can be sure that the components aren't an issue.

I'll update tomorrow.

consolacion, 2017-04-11:
Thanks. good luck. I will be checking this space.
Working on some IR projects as wel, about to order several IR receivers, hence my interest

mihalski, 2017-04-12:
I had a working theory that perhaps even when idle and unused my Oculus Rift HMD was leaking IR and causing the IR receiver to trigger with random input. Unfortunately that doesn't not seem to be the case. Even TOTALLY unpowered I get the same behaviour.

consolacion, 2017-04-13:
That's too bad. Hmm, back to the drawing board

mihalski, 2017-04-13:
I did however receive the AnalysIR gear today and hope to play with it during the weekend. Maybe it will help clarify what is going on.

consolacion, 2017-04-13:
Always good to get a package :-) I hope you will find a solution.
I have gone ahead and ordered the TSOP38238

consolacion, 2017-04-14:
Not sure if this will solve your problem, but while waiting for my TSOP's I did some further playing with an old SFH (an 5 Volt SFH 5110 or something). I noticed that when I keep it close to the ESP8266 chip, I seem to get some random values. When I keep it  on a 20 cm dupont cable away from my Wemos there are no problems

mihalski, 2017-04-14:
I will look into that. I recently discovered that the same is true of most (all) analogue PIR sensors. There is at least one which uses a DSP IC that is supposed to be immune and I have ordered some for testing.

mihalski, 2017-04-14:
But I just realised that I have also tried with non-wifi, non-esp8266 MCU's and had the same problem. Still, I will keep it under consideration.

consolacion, 2017-04-14:
You probably mean this one: https://www.aliexpress.com/item/New-Arrival-Mini-IR-Pyroelectric-Infrared-PIR-Motion-Human-Sensor-Automatic-Detector-Module-high-reliability-12mm/32749804501.html?spm=2114.13010308.0.0.zR4zyh
ordered that too
Hmm, too bad abt that. Obviously another cause

mihalski, 2017-04-14:
Indeed.. I ordered 12.

consolacion, 2017-04-14:
12.  You have big plans :-)
I guess you have considered the usual suspects, but heatsources, maybe even a wireless phone or router if close could be a culprit. I noticed that if I hold my SFHxxxx sturdy between my fingers (casing only) I get random codes. Is there maybe another ESP in the vicinity?
Now ofcourse  an IR module shld be ok  without you worrying about all those things but switching any possible electric device off might be  a start to locate the problem.
Having it breadboarded can also be a source of problems. You may want to try a small ferrite bead in the Vcc, close to the module and a 4.7uF + 100nF behind that. 
Theoretically ofcourse the 100nF parallel to a 4.7uF seems ludicrous but some people swear by it to get rid of high frequency garble.
I hope your AnalysIR goods got you in the right direction.
Anyway, summarizing.. my SFH5110-38, works virtually error free, (but it is a 5Volt and I want to go to a 3.3 V) and I see no reason why the TSOP34838 shouldnt work error free. Telltale is that you mentioned it doesnt work on 3.3 Volt, whereas according to datasheet it should work from 2.5. I am sure you are capable enough to wire the thing correctly, but  if you havent, make sure you have that 4.7uF capacitor  over its supply line, just to catch possible short voltage drops

mihalski, 2017-04-19:
Well I can tell you this. The LearnIR module does not see ANY of the random IR stuff I was picking up with my DIY barebones module. I'll eventually figure this out. Only get to mess with it sporadically though.

consolacion, 2017-04-19:
Good to know. kinda suggests a component or wiring issue. Anyway, I am happy you got a lead

closed 2017-06-02 by mdhiggins