mdhiggins/ESP8266-HTTP-IR-Blaster #23: Error when uploading Blueprint

jacksaunders, 2017-04-07:
I am getting this error when attempting to upload the blueprint to the board.

```
/Users/***/Downloads/ESP8266-HTTP-IR-Blaster-master/IRController/IRController.ino:6:99: fatal error: WiFiManager.h: No such file or directory
 #include <WiFiManager.h>          // https://github.com/tzapu/WiFiManager WiFi Configuration Magic
                                                                                                   ^
compilation terminated.
Multiple libraries were found for "WiFiUdp.h"
 Used: /Users/****/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Error compiling for board Generic ESP8266 Module.
```

Hope someone can help thanks

mdhiggins, 2017-04-07:
This is related to the new update. You need to install Wifi Manager in the arduino library. I'll update the readme.

mdhiggins, 2017-05-12:
Readme updated

closed 2017-05-12 by mdhiggins