mdhiggins/ESP8266-HTTP-IR-Blaster #25: unnable to connect to the esp

saikhurana, 2017-05-19:
the device goes into the wifi configuration mode is successfully able to connect to my home network but when i enter the url provided in the serial monitor, it is not able to connect to the http server. But is says server is running when i enter the ip address.

mdhiggins, 2017-05-19:
To clarify then the IP address works but the more friendly http://device.local doesn't work?

saikhurana, 2017-05-19:
yeah. that's exactly what i am trying to say

mdhiggins, 2017-05-19:
That can sometimes be an issue with your router. It's using bonjour/mdns so make sure those services are compatible and enabled

saikhurana, 2017-05-19:
can you please help me to find this option. i am not familiar with this networking stuff

saikhurana, 2017-05-19:
Is there any workaround

mdhiggins, 2017-05-19:
Using the IP address is a good alternative. 

Otherwise I'd need to know your OS and router information

saikhurana, 2017-05-19:
i use win10 and i have 2 routers. d-link and a binatone.
d-link DSL-2750U Wireless N 300 ADSL2
Binatone WR1500N

saikhurana, 2017-05-19:
I am fine using the ip address. Can you explain the procedure

saikhurana, 2017-05-20:
how to connect to the ir remote via ip adress. when i type the local ip address of the it, it just says "Server is running"

mdhiggins, 2017-05-20:
See the readme for the URL formatting. It's all spelled out very clearly there

saikhurana, 2017-05-21:
is there a GUI when you use the local host method? is it accessible using the ip address

mdhiggins, 2017-05-21:
No there is no GUI accept for the initial wifi setup everything else is handled via URL parameters

closed 2017-05-22 by saikhurana