mdhiggins/ESP8266-HTTP-IR-Blaster #27: Sample URL request

saikhurana, 2017-06-02:
hey there.
you have an amazing code here. i am not very good at programming and in networking stuff.
i want few example urls that will with ifttt
1. sending raw data.
2. sending encoded data
3. sending different ir raw signal. one followed by the other for a certain talk(eg turn on the tv and then press the source button 2 times.)

please keep in the mind that the url should work with ifttt.
thanks in advance

i apologize for my poor English in advance

mdhiggins, 2017-06-02:
There are examples to all of these things in the readme (with the exception of of number 3 specifically)

Everything in the readme is designed to work with IFTTT

I will post some examples below however

1. `http://xxx.xxx.xxx.xxx:xxxx/json?plain=[{"type":"raw","data":[2450,600, 1300,600, 700,550, 1300,600, 700,550, 1300,550, 700,550, 700,600, 1300,600, 700,550, 700,550, 700,550, 700],"khz":38,"pulse":3}]&pass=yourpass`

2. `http://xxx.xxx.xxx.xxx:xxxx/msg?code=A90:SONY:12&pulse=2&repeat=5&pass=yourpass`

3. `http://xxx.xxx.xxx.xxx:xxxx/json?plain=[{"type":"raw","data":[2450,600, 1300,600, 700,550, 1300,600, 700,550, 1300,550, 700,550, 700,600, 1300,600, 700,550, 700,550, 700,550, 700],"khz":38,"},{"type":"raw","data":[2450,600, 1300,600, 700,550, 1300,600, 700,550, 1300,550, 700,550, 700,600, 1300,600, 700,550, 700,550, 700,550, 700],"khz":38,"repeat":2}]&pass=yourpass`

The raw data used in these examples is random and doesn't reflect your power button or source button, you'll need to capture that data and exchange it. And obviously the `xxx.xxx.xxx.xxx:xxxx` needs to be replaced with your external IP address and port

closed 2017-06-02 by mdhiggins

saikhurana, 2017-06-03:
hey if i want to change the IR port then what will be the url.
please give example for both encoded and raw signals.