mdhiggins/ESP8266-HTTP-IR-Blaster #28: Ir send port change issure

saikhurana, 2017-06-03:
Hey there i want to change the preset gpio pins fir sending ir signals as gpio 13 and 0 do not work. 
When i changed the ir pins in the. The esp8266 connects to the internet but the line "ready to send and recieve ir signals" does not show after the change and the module reconnect over and over again.

mdhiggins, 2017-06-03:
GPIO13 is reserved for resetting the wifi configuration settings. If you have hardware plugged in to this pin it will likely just keep resetting. I would use a different pin or change the reset pin in the blueprint.

saikhurana, 2017-06-03:
i tried to change the pin to gpio6 but the esp8266 keeps on resetting itself. this doesn't happen when i used the default values 

in the cases, gpio 13 is not connected to anything

saikhurana, 2017-06-03:
also please give me an example URL if i want to transmit my ir signal through a different ir led as Defined in the Code

mdhiggins, 2017-06-03:
There is an example in the readme for these settings

closed 2017-06-11 by mdhiggins