mdhiggins/ESP8266-HTTP-IR-Blaster #30: compilations problems

roybm, 2017-06-15:
I went step by step after the guide
and stuck for a while in the upload step because of unsuccessful compilation.
first I got errors -

**********************************************************
Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

IRController:32: error: 'IRrecv' does not name a type

 IRrecv irrecv(5);                 // Receiving pin (GPIO5 = D1)

 ^

IRController:33: error: 'IRsend' does not name a type

 IRsend irsend1(4);                // Transmitting preset 1

 ^

IRController:34: error: 'IRsend' does not name a type

 IRsend irsend2(0);                // Transmitting preset 2

 ^

IRController:35: error: 'IRsend' does not name a type

 IRsend irsend3(12);               // Transmitting preset 3

 ^

IRController:36: error: 'IRsend' does not name a type

 IRsend irsend4(13);               // Transmitting preset 4

 ^

IRController:367: error: variable or field 'ircode' declared void

 void  ircode (decode_results *results)

             ^

IRController:367: error: 'decode_results' was not declared in this scope

IRController:367: error: 'results' was not declared in this scope

 void  ircode (decode_results *results)

                             ^

IRController:383: error: 'IRsend' does not name a type

 IRsend pickIRsend (int out) {

 ^

IRController:397: error: variable or field 'encoding' declared void

 void  encoding (decode_results *results)

               ^

IRController:397: error: 'decode_results' was not declared in this scope

IRController:397: error: 'results' was not declared in this scope

 void  encoding (decode_results *results)

                               ^

IRController:422: error: variable or field 'fullCode' declared void

 void fullCode (decode_results *results)

               ^

IRController:422: error: 'decode_results' was not declared in this scope

IRController:422: error: 'results' was not declared in this scope

 void fullCode (decode_results *results)

                               ^

IRController:437: error: variable or field 'dumpInfo' declared void

 void dumpInfo (decode_results *results)

               ^

IRController:437: error: 'decode_results' was not declared in this scope

IRController:437: error: 'results' was not declared in this scope

 void dumpInfo (decode_results *results)

                               ^

IRController:456: error: variable or field 'dumpRaw' declared void

 void  dumpRaw (decode_results *results)

              ^

IRController:456: error: 'decode_results' was not declared in this scope

IRController:456: error: 'results' was not declared in this scope

 void  dumpRaw (decode_results *results)

                              ^

IRController:487: error: variable or field 'dumpCode' declared void

 void  dumpCode (decode_results *results)

               ^

IRController:487: error: 'decode_results' was not declared in this scope

IRController:487: error: 'results' was not declared in this scope

 void  dumpCode (decode_results *results)

                               ^

IRController:564: error: 'IRsend' has not been declared

 void irblast(String type, String dataStr, int len, int rdelay, int pulse, int pdelay, int repeat, long address, IRsend irsend) {

                                                                                                                 ^

IRController:610: error: 'IRsend' has not been declared

 void rawblast(JsonArray &raw, int khz, int rdelay, int pulse, int pdelay, int repeat, IRsend irsend) {

                                                                                       ^

IRController:633: error: 'IRsend' has not been declared

 void roomba_send(int code, int pulse, int pdelay, IRsend irsend)

                                                   ^

C:\Users\roiyb\Desktop\IRController\IRController.ino: In lambda function:

IRController:265: error: 'pickIRsend' was not declared in this scope

           rawblast(raw, khz, rdelay, pulse, pdelay, repeat, pickIRsend(out));

                                                                           ^

IRController:273: error: 'pickIRsend' was not declared in this scope

           irblast(type, data, len, rdelay, pulse, pdelay, repeat, address, pickIRsend(out));

                                                                                          ^

C:\Users\roiyb\Desktop\IRController\IRController.ino: In lambda function:

IRController:307: error: 'pickIRsend' was not declared in this scope

         irblast(type, data, len, rdelay, pulse, pdelay, repeat, address, pickIRsend(out));

                                                                                        ^

C:\Users\roiyb\Desktop\IRController\IRController.ino: In function 'void setup()':

IRController:321: error: 'irsend1' was not declared in this scope

   irsend1.begin();

   ^

IRController:322: error: 'irsend2' was not declared in this scope

   irsend2.begin();

   ^

IRController:323: error: 'irsend3' was not declared in this scope

   irsend3.begin();

   ^

IRController:324: error: 'irsend4' was not declared in this scope

   irsend4.begin();

   ^

IRController:325: error: 'irrecv' was not declared in this scope

   irrecv.enableIRIn();

   ^

C:\Users\roiyb\Desktop\IRController\IRController.ino: At global scope:

IRController:367: error: variable or field 'ircode' declared void

 void  ircode (decode_results *results)

               ^

IRController:367: error: 'decode_results' was not declared in this scope

IRController:367: error: 'results' was not declared in this scope

 void  ircode (decode_results *results)

                               ^

Multiple libraries were found for "WiFiUdp.h"
 Used: C:\Users\roiyb\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'IRrecv' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
  

I tried to solve it in several ways (like adding  #include <IRrecv.h> or reinstalling the library... but it made other problems and I fill it is not the way.

I will be happy for some help here
thanks

dham1012, 2017-06-15:
I am getting very similar results. I have added a number of includes to stop the IRrecv issues. Errors I am getting are below

Arduino: 1.8.3 (Windows Store 1.8.6.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\.....\Documents\Arduino\IRController\IRController.ino: In function 'void ircode(decode_results*)':

IRController:372: error: 'class decode_results' has no member named 'panasonicAddress'

     Serial.print(results->panasonicAddress, HEX);

                           ^

IRController:377: error: call of overloaded 'print(uint64_t&, int)' is ambiguous

   Serial.print(results->value, HEX);

                                   ^

C:\Users\.....\Documents\Arduino\IRController\IRController.ino:377:35: note: candidates are:

In file included from C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Stream.h:26:0,

                 from C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/HardwareSerial.h:31,

                 from C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:245,

                 from sketch\IRController.ino.cpp:1:

C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.h:71:16: note: size_t Print::print(unsigned char, int)

         size_t print(unsigned char, int = DEC);

                ^

C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.h:72:16: note: size_t Print::print(int, int)

         size_t print(int, int = DEC);

                ^

C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.h:73:16: note: size_t Print::print(unsigned int, int)

         size_t print(unsigned int, int = DEC);

                ^

C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.h:74:16: note: size_t Print::print(long int, int)

         size_t print(long, int = DEC);

                ^

C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.h:75:16: note: size_t Print::print(long unsigned int, int)

         size_t print(unsigned long, int = DEC);

                ^

C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.h:76:16: note: size_t Print::print(double, int)

         size_t print(double, int = 2);

                ^

C:\Users\.....\Documents\Arduino\IRController\IRController.ino: In function 'void fullCode(decode_results*)':

IRController:426: error: call of overloaded 'print(uint64_t&, int)' is ambiguous

   Serial.print(results->value, HEX);

[... 115 lines cut ...]

C:\Users\.....\Documents\Arduino\IRController\IRController.ino:658:31: note: candidate is:

In file included from C:\Users\.....\Documents\Arduino\IRController\IRController.ino:8:0:

C:\Users\.....\Documents\Arduino\libraries\IRremoteESP8266\src/irsend.h:37:8: note: void IRsend::sendRaw(uint16_t*, uint16_t, uint16_t)

   void sendRaw(uint16_t buf[], uint16_t len, uint16_t hz);

        ^

C:\Users\.....\Documents\Arduino\libraries\IRremoteESP8266\src/irsend.h:37:8: note:   no known conversion for argument 1 from 'unsigned int [(((sizetype)(((ssizetype)(length * 2)) + -1)) + 1)]' to 'uint16_t* {aka short unsigned int*}'

Multiple libraries were found for "WiFiUdp.h"
 Used: C:\Users\.....\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.6.0_x64__mdqgnx93n4wtt\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.6.0_x64__mdqgnx93n4wtt\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.6.0_x64__mdqgnx93n4wtt\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.6.0_x64__mdqgnx93n4wtt\libraries\WiFi
exit status 1
'class decode_results' has no member named 'panasonicAddress'

Would really appreciate some steer to fix, its a great project.

If i comment the lines out i can get it to compile, but suspect im missing stuff at this point

mdhiggins, 2017-06-15:
Did you install all the required libraries including the IR library which I believe has to be manually installed?

b0bje, 2017-06-15:
I added the below lines and a great deal of the errors disapear (as
dham1012 recommended)

#include <IRsend.h>
#include <IRrecv.h>

But that indeed leaves the other errors like:

'class decode_results' has no member named 'panasonicAddress'

I'm guessing it is due to the 2.0 release of those esp8266IR libraries..

dham1012, 2017-06-15:
@b0bje @roybm I have just changed the version of IRremoteESP8266 libraries installed to 1.2.0, seems to compile with no errors. I removed the additional #include.
I haven't had the chance to load it, verifies ok.

dham1012, 2017-06-15:
Have now loaded it and all appears to be working g as expected. 
@mdhiggins, any thoughts on the changes to this library? Maybe worth stating the versions you installed, just an idea? 
Great work btw,  I'm a total novice at this stuff and this is exactly what I was looking for, thank you. 
While I'm here, does it matter on the value of  resistors used, again this isn't my thing and have totaly guessed based on matching the colours of the resistors in your image, though the second one is hard to see!

mdhiggins, 2017-06-15:
Yeah looks like this is all related to the 2.0 release, use the older versions for now and I"ll have to work on an update

mdhiggins, 2017-06-15:
@dham1012  see https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/12 for resistor information

dham1012, 2017-06-15:
@mdhiggins, perfect, thanks.

roybm, 2017-06-15:
great , thanks.

mdhiggins, 2017-06-18:
Just pushed an update that makes the project compatible with version 2.0 should you want to use the newer features
New version unfortunately isn't backwards compatible but should at least behave going forward with future versions

roybm, 2017-06-18:
Thanks for the update.
Now dealing with some new compilation issues maybe someone can help?
****************************************************
Build options changed, rebuilding all
.....

IRController:380: error: 'serialPrintUint64' was not declared in this scope

   serialPrintUint64(results->value, 16);

                                       ^

C:\Users\roiyb\Downloads\IRController\IRController.ino: In function 'void fullCode(decode_results*)':

IRController:429: error: 'serialPrintUint64' was not declared in this scope

   serialPrintUint64(results->value, 16);

                                       ^

C:\Users\roiyb\Downloads\IRController\IRController.ino: In function 'void dumpCode(decode_results*)':

IRController:513: error: 'serialPrintUint64' was not declared in this scope

   serialPrintUint64(results->value, 16);

                                       ^

Multiple libraries were found for "WiFiUdp.h"
 Used: C:\Users\roiyb\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'serialPrintUint64' was not declared in this scope

********************************************

mdhiggins, 2017-06-18:
Do you have `#include <IRutils.h>` in your headers?

roybm, 2017-06-19:
yes...

dham1012, 2017-06-19:
Hi, some feedback from me. 
I updated yesterday, all compiled and loaded fine here, I did a full copy and paste over all my other code and updated the IRremoteESP8266 library to 2.0.3. Seems to be fine sending, but when I received IR the results were wrong, for example, it didn't recognise my panasonic remote, even sometimes coming back as sony etc. Code were received, but the received codes were not correct.
I rolled back to previous version of the code and library and tried again, codes all received and recognised fine.

roybm, 2017-06-19:
Working for me only for IRremoteESP8266  version 2.0.2.

closed 2017-06-19 by roybm

mdhiggins, 2017-06-19:
@roybm what versions was it failing with? (shouldn't work on 1.x.x versions)

mdhiggins, 2017-06-19:
Gonna push an update for some of the serial output to make sure there wasn't additional changes with the 2.x updates @dham1012, I'll let you know if may be screwing up the serial output for reads

dham1012, 2017-06-19:
Ok thanks, will give it a test when you say.

mdhiggins, 2017-06-19:
Just pushed a series of updates that should address the issue. It looks like there was some known issues with panasonic codes that required an update. Hopefully fixed now (though I don't have any panasonic devices to test with)

Additionally, some new features!
Going to the web address root that would previously just say "Server is running" now gives you some device stats.
Going to http://device.local/last will now show you your last received IR code so you can grab new codes without needing to be connected to serial monitoring

dham1012, 2017-06-19:
Fantastic, trying now.
I'm current in the process of integrating this with SmartThings to do the interface between Alexa and the ESP, so far so good. Great piece of work this.

mdhiggins, 2017-06-19:
Hm I would post your issue about the panasonic codes on https://github.com/markszabo/IRremoteESP8266 and see if he has any suggestions, I'm using his exact implementation of the code so if there's a problem and its not recognizing panasonic it may be related to changes to the library itself

dham1012, 2017-06-21:
Had a look at the issues section in https://github.com/markszabo/IRremoteESP8266, it seems there was a reported issue with Panasonic remotes, indicates was fixed in version 2.0.2 of the library, but I'm sure I was using that. I will try again and report there if no success, thanks for the pointer.