mdhiggins/ESP8266-HTTP-IR-Blaster #31: I have a couple of questions

NorthernMan54, 2017-06-18:
First, great project worked first time.

1) The response time seems really slow, i.e. 7 seconds for an action to occur.  is this normal?

2) When I captured my LG TV and Yamaha Receiver the codes looked like this

echo "LG On"
curl http://irblaster.local/msg?code=20DF23DC:NEC:32

echo "LG Off"
curl http://irblaster.local/msg?code=20DFA35C:NEC:32

echo "Yamaha On"
curl http://irblaster.local/msg?code=5EA1B847:NEC:32

echo "Yamaha Off"
curl http://irblaster.local/msg?code=5EA17887:NEC:32

mdhiggins, 2017-06-18:
7 seconds is a long time, but that's probably related either to IFTTT or your internet connection, the response time of the local device should be closer to 1 second
You can test this by executing a command from your browser using your local address

Is number 2 a question? Those CURL URLs look good otherwise (though you may need to include a pass if you set one). Also make sure you specify your port unless you used port 80

NorthernMan54, 2017-06-18:
Tks Michael

The 7 seconds was local.  I was using my mac and the curl command nothing fancy.  Tks for confirming that something isn't right with my setup.

NorthernMan54, 2017-06-20:
I found my response time issue.  It is something to do with mdns name resolution on my Mac.

If I use the mdns version of the address i.e. http://irblaster.local/msg?code=A90:SONY:12 it takes 6 to 7 seconds.  If I use the IP address it takes less than a second.

When I tried from Raspberry PI, I didn't have this issue.   You learn something new every day.

closed 2017-06-20 by NorthernMan54