mdhiggins/ESP8266-HTTP-IR-Blaster #33: post call

arielelevy, 2017-06-27:
I can't do /json work calling by POST

the  json content is read from the "plain" arg, and regarding ESP8266WebServer source code, plain is the default arg when the post doesn't have the content-type header or the content-type is "text/plain"

My call is like this:

POST /json?pass=1234
Content-Length: 110

[{"type":"raw","data":[5900,1300,1150,1250,650,600,650,600,650,600,1150,1250,650,1200,1150,650,1150,650,650]}]

NorthernMan54, 2017-06-27:
FYI - This works in a browser window

http://ESP_869815/json?plain=[{%22type%22:%22raw%22,%22data%22:[3428,1767,428,428,428,1312,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,1312,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,1312,428,428,428,1312,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,428,1312,428,1312,428,1312,428,1312,428,1312,428,428,428,428,428,428,428,1312,428,1312,428,1312,428,1312,428,428,428,428,428,1312,428],%22khz%22:38,%22pulse%22:1}]

mdhiggins, 2017-06-27:
Yeah you can use the browser or just set your content-type to "text/plain"
This was done to have it place nice with IFTTT

arielelevy, 2017-06-28:
Hi! do you test it calling by POST instead of GET?
Not by adding argument at URI, but calling from Chrome "advanced rest client" as example, POST allows unlimited messages sizes.

mdhiggins, 2017-06-28:
I tested using POST with cURL mostly

arielelevy, 2017-06-28:
and body just the json? 

[{"type":"raw","data":[5900,1300,1150,1250,650,600,650,600,650,600,1150,1250,650,1200,1150,650,1150,650,650]}]

mdhiggins, 2017-06-28:
I believe so, yes

arielelevy, 2017-06-28:
Greate work! 

I will try POST, later... I'm doing something wrong

As example of raw, a sony call, 3 minimum pulses, 45 microsec between them, and 40Khz signal. Works! 

http://192.168.1.13/json?pass=1234&plain=[{"type":"raw","data":[2400,600,1200,600,1200,600,600,600,600,600,1200,600,600,600,600,600,600,600,600,600,600,600,600,600,1200,600,1200,600,600,600,600],"khz":40,"pdelay":45,"pulse":3}]

arielelevy, 2017-06-28:
I got working by POST.

POST /json?pass=1234 HTTP/1.10x0D0x0AHost: 192.168.1.130x0D0x0AAccept: */*0x0D0x0AContent-Length: 1970x0D0x0A
Content-Type: text/plain0x0D0x0A0x0D0x0A[{"type":"raw","data":[2400,600,1200,600,1200,600,600,600,600,600,1200,600,600,600,600,600,600,600,600,600,600,600,600,600,1200,600,1200,600,600,600,600],"khz":40,"pdelay":45,"pulse":3,"repeat":2}]0x0D0x0A0x0D0x0A

arielelevy, 2017-06-28:
simplest... only content length is required.

POST /json?pass=1234 HTTP/1.10x0D0x0AContent-Length: 1970x0D0x0A0x0D0x0A[{"type":"raw","data":[2400,600,1200,600,1200,600,600,600,600,600,1200,600,600,600,600,600,600,600,600,600,600,600,600,600,1200,600,1200,600,600,600,600],"khz":40,"pdelay":45,"pulse":3,"repeat":2}]0x0D0x0A0x0D0x0A

closed 2017-06-28 by arielelevy

mdhiggins, 2017-06-28:
Thanks for sharing