mdhiggins/ESP8266-HTTP-IR-Blaster #34: Last IR code received

robste11159, 2017-06-29:
I tried to use the /last command last night but it was not recognised. Bug?
PS: great project!

mdhiggins, 2017-06-29:
Can you be more specific?
Did you just get a blank page? Were you getting anything on the serial output to indicate the code was received?

If you're seeing the codes pop up on the serial output but the page isn't showing you anything then that would be an issue

robste11159, 2017-06-29:
If I browse to http://192.168.2.10:8080/
Server is running

If I browse to http://192.168.2.10:8080/last
Not found: /last

mdhiggins, 2017-06-29:
Sounds like you didn't update to the latest version if all you see is "Server is running"

The /last feature was added a few weeks ago and the root page was updated with a bunch of device info.

mdhiggins, 2017-07-01:
Gonna close this since there's been a bunch of updates and the /last page is no longer in existence

closed 2017-07-01 by mdhiggins