mdhiggins/ESP8266-HTTP-IR-Blaster #35: Force WiFi reconfiguration not working

jimbroze, 2017-06-29:
I had some issues connecting my NodeMCU to my router but after rebooting the chip it would neither connect to the router or create an access point. Forcing a reconfig by grounding GPIO13 provides the following serial message upon startup:

`ESP8266 IR Controller
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
mounted file system
`
The ModeMCU now provides an access point if I flash old firmware but  not when using Arduino IDE and WiFIManager. Despite the output suggesting that it is doing so:
`ESP8266 IR Controller
mounted file system
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
Entered config mode
0.0.0.0
IRBlaster Configuration
*WM: 
*WM: Configuring access point... 
*WM: IRBlaster Configuration
*WM: AP IP address: 
*WM: 192.168.244.1
*WM: HTTP server started
`

mdhiggins, 2017-06-29:
When you say flashing old firmware what version specifically are you referring to?
What version of WiFi Manager is giving your trouble?

jimbroze, 2017-06-29:
The last version that was released with NodeMCU flasher: https://github.com/nodemcu/nodemcu-flasher.
It's extremely old and was used just to check that the wifi hardware was not broken. I've also tried flashing the latest version of ESP8266 Basic which doesn't provide working WiFi either.
I'm using the latest version of WiFi Manager (0.12) but haven't tried any others .

mdhiggins, 2017-06-29:
If you're using this blueprint and the arduino you shouldn't be using nodemcu flasher or any custom firmware, the arduino blueprint contains all the firmware and anything you flash prior to uploading the blueprint is overwritten.

The nodemcu flasher is for if you want to use the LUA language which is completely different from flashing it with the arduino firmware

mdhiggins, 2017-06-29:
Also ESP8266Basic is again another firmware that is different from the previously mentioned 2 firmwares

If you're getting no WiFi on all 3 firmware devices then I would look into your hardware or router having an issue. But just wanted to clarify that you can't run all 3 of these simultaneously

jimbroze, 2017-06-29:
That's why I was just using it to rule out a hardware problem. I can create an access point using the nodemcu flasher firmware but get the above issues when using your blueprint and arduino for some reason

jimbroze, 2017-06-29:
Also, after flashing the NodeMCU flasher firmware, the access point name still registers as "IRBlaster Configuration". Could the issue be caused by credentials or other data being saved despite the re-flash?

mdhiggins, 2017-06-29:
I believe the ESP8266 Basic flasher has the ability to format all the filesystem data, you could try formatting that data an then refreshing the firmware and see if that persists

After flashing with the nodeMCU flasher what sort of serial output are you seeing? I would post all of that information after the nodeMCU flash

jimbroze, 2017-06-29:
Using the nodeMCU flasher gives the following output:
[image]
And creates an access point called "IRBlaster Configuration". Unfortunately the format function of the ESP8266Basic flasher crashes when I try to use it.

mdhiggins, 2017-06-29:
Actually now that I'm thinking about it, the SSID of the WiFi network is stored in the blueprint and not written to the filesystem at all, so formatting the device wouldn't help anyway.

I'm thinking the nodeMCU flasher probably isn't actually working and you're just continuing to the run the Arduino blueprint as you would expect.

What happens when you attempt to connect to the IRBlaster Configuration SSID? Keep an eye on the serial output when you connect and try to navigate to 192.168.244.1

jimbroze, 2017-06-29:
Strangely, I can connect to the access point but there quite a large delay and also nothing is outputted to the serial monitor. Attempting to connect to the IP reaches a timeout and once again does not output anything to the serial monitor.

jimbroze, 2017-06-29:
Strangely, I can connect to the access point but there is a large delay and
nothing is output from the serial monitor. Attempting to connect to the IP
reaches a timeout and once again nothing is outputted.

mdhiggins, 2017-06-29:
Hm this is all very strange and I'm unable to reproduce what's happening on your end, tried on multiple NodeMCU devices, wonder if maybe you've got a bad board/wifi chip?

If you do a clean install and force setup mode, you still cannot connect?

jimbroze, 2017-06-29:
I'm starting to think that's the only possibility! Unfortunately I only have the one board but I'll try and get hold of another board from a different supplier.

Do you mean a clean install of Arduino?

mdhiggins, 2017-07-01:
Yeah you could try clearing out your arduino library and reinstall all your dependencies to see if that makes a difference

Would also try changing the SSID of the configuration hotspot and see if that makes a difference

closed 2017-07-09 by mdhiggins