mdhiggins/ESP8266-HTTP-IR-Blaster #36: Interface

dham1012, 2017-06-29:
@mdhiggins , I've created an interface for /, not sure if you are interested in including. I'm not that familiar to how github works in this respect!

[image]

mdhiggins, 2017-06-29:
Wow that's really nice looking, yes please share

dham1012, 2017-06-29:
Not quite sure how, paste .ino code here?

mdhiggins, 2017-06-29:
You can either do a pull request or just paste it here and I'll integrate it

dham1012, 2017-06-29:
I'll paste! It uses the bootstrap css stuff, which is referenced.

I havent included the most recent changes you have made, such as the external IP lookup.

dham1012, 2017-06-29:
`#include <FS.h>                   // This needs to be first, or it all crashes and burns

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <DNSServer.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>          // https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <WiFiUdp.h>
#include <ESP8266mDNS.h>          // useful to access to ESP by hostname.local
#include <NTPClient.h>
#include <ArduinoJson.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>

#include <Ticker.h>               // For LED status

const int configpin = 13;         // GPIO13 (D7 on D1 Mini) to enable configuration (connect to ground)
const char *wifi_config_name = "IRBlaster Configuration";
int port = 80;
char passcode[40] = "ESP9951012";
char host_name[40] = "ESPIRremote";
String last_code = "";            // Stores last code
String last_code_2 = "";          // Stores 2nd to last code
String last_code_3 = "";          // Stores 3rd to last code
String last_code_short = "";            // Stores last code
String last_code_short_2 = "";          // Stores 2nd to last code
String last_code_short_3 = "";          // Stores 3rd to last code
String last_send_time = "";            // Stores last send code
String last_send_data = "";            // Stores last send code
String last_send_type = "";            // Stores last send code
String last_send_address = "";            // Stores last send code
String last_send_time_2 = "";            // Stores last send code
String last_send_data_2 = "";            // Stores last send code
String last_send_type_2 = "";            // Stores last send code
String last_send_address_2 = "";            // Stores last send code
String last_send_time_3 = "";            // Stores last send code
String last_send_data_3 = "";            // Stores last send code
String last_send_type_3 = "";            // Stores last send code
String last_send_address_3 = "";            // Stores last send code

String last_code_time = "";            // Stores last send code
String last_code_command = "";            // Stores last send code
String last_code_type = "";            // Stores last send code
String last_code_length = "";
String last_code_address = "";            // Stores last send code
String last_code_time_2 = "";            // Stores last send code
String last_code_command_2 = "";            // Stores last send code
String last_code_type_2 = "";            // Stores last send code
String last_code_length_2 = "";
String last_code_address_2 = "";            // Stores last send code
String last_code_time_3 = "";            // Stores last send code
String last_code_command_3 = "";            // Stores last send code
String last_code_type_3 = "";            // Stores last send code
String last_code_length_3 = "";
String last_code_address_3 = "";            // Stores last send code

ESP8266WebServer server(port);
HTTPClient http;
Ticker ticker;

WiFiUDP ntpUDP;

// You can specify the time server pool and the offset (in seconds, can be
// changed later with setTimeOffset() ). Additionaly you can specify the
// update interval (in milliseconds, can be changed using setUpdateInterval() ).
NTPClient timeClient(ntpUDP, "europe.pool.ntp.org", 3600, 60000);

bool shouldSaveConfig = false;    // Flag for saving data
[... 831 lines cut ...]
//    last_code_short = codeOutputSRT(&results);

    last_code_time_3 = last_code_time_2;
    last_code_command_3 = last_code_command_2;
    last_code_type_3 = last_code_type_2;
    last_code_length_3 = last_code_length_2;
    last_code_address_3 = last_code_address_2;
    
    last_code_time_2 = last_code_time;
    last_code_command_2 = last_code_command;
    last_code_type_2 = last_code_type;
    last_code_length_2 = last_code_length;    
    last_code_address_2 = last_code_address;
    
    last_code_time = codeOutputSRT(int(1), &results);
    last_code_command = codeOutputSRT(int(2), &results);
    last_code_type = codeOutputSRT(int(3), &results);
    last_code_length = codeOutputSRT(int(4), &results);
    last_code_address = codeOutputSRT(int(5), &results);    
  
    Serial.println("");           // Blank line between entries
    irrecv.resume();              // Prepare for the next value
  }
  delay(200);
}`

dham1012, 2017-06-29:
#include <FS.h>                   // This needs to be first, or it all crashes and burns

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <DNSServer.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>          // https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <WiFiUdp.h>
#include <ESP8266mDNS.h>          // useful to access to ESP by hostname.local
#include <NTPClient.h>
#include <ArduinoJson.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>

#include <Ticker.h>               // For LED status

const int configpin = 13;         // GPIO13 (D7 on D1 Mini) to enable configuration (connect to ground)
const char *wifi_config_name = "IRBlaster Configuration";
int port = 80;
char passcode[40] = "ESP9951012";
char host_name[40] = "ESPIRremote";
String last_code = "";            // Stores last code
String last_code_2 = "";          // Stores 2nd to last code
String last_code_3 = "";          // Stores 3rd to last code
String last_code_short = "";            // Stores last code
String last_code_short_2 = "";          // Stores 2nd to last code
String last_code_short_3 = "";          // Stores 3rd to last code
String last_send_time = "";            // Stores last send code
String last_send_data = "";            // Stores last send code
String last_send_type = "";            // Stores last send code
String last_send_address = "";            // Stores last send code
String last_send_time_2 = "";            // Stores last send code
String last_send_data_2 = "";            // Stores last send code
String last_send_type_2 = "";            // Stores last send code
String last_send_address_2 = "";            // Stores last send code
String last_send_time_3 = "";            // Stores last send code
String last_send_data_3 = "";            // Stores last send code
String last_send_type_3 = "";            // Stores last send code
String last_send_address_3 = "";            // Stores last send code

String last_code_time = "";            // Stores last send code
String last_code_command = "";            // Stores last send code
String last_code_type = "";            // Stores last send code
String last_code_length = "";
String last_code_address = "";            // Stores last send code
String last_code_time_2 = "";            // Stores last send code
String last_code_command_2 = "";            // Stores last send code
String last_code_type_2 = "";            // Stores last send code
String last_code_length_2 = "";
String last_code_address_2 = "";            // Stores last send code
String last_code_time_3 = "";            // Stores last send code
String last_code_command_3 = "";            // Stores last send code
String last_code_type_3 = "";            // Stores last send code
String last_code_length_3 = "";
String last_code_address_3 = "";            // Stores last send code

ESP8266WebServer server(port);
HTTPClient http;
Ticker ticker;

WiFiUDP ntpUDP;

// You can specify the time server pool and the offset (in seconds, can be
// changed later with setTimeOffset() ). Additionaly you can specify the
// update interval (in milliseconds, can be changed using setUpdateInterval() ).
NTPClient timeClient(ntpUDP, "europe.pool.ntp.org", 3600, 60000);

bool shouldSaveConfig = false;    // Flag for saving data
[... 831 lines cut ...]
//    last_code_short = codeOutputSRT(&results);

    last_code_time_3 = last_code_time_2;
    last_code_command_3 = last_code_command_2;
    last_code_type_3 = last_code_type_2;
    last_code_length_3 = last_code_length_2;
    last_code_address_3 = last_code_address_2;
    
    last_code_time_2 = last_code_time;
    last_code_command_2 = last_code_command;
    last_code_type_2 = last_code_type;
    last_code_length_2 = last_code_length;    
    last_code_address_2 = last_code_address;
    
    last_code_time = codeOutputSRT(int(1), &results);
    last_code_command = codeOutputSRT(int(2), &results);
    last_code_type = codeOutputSRT(int(3), &results);
    last_code_length = codeOutputSRT(int(4), &results);
    last_code_address = codeOutputSRT(int(5), &results);    
  
    Serial.println("");           // Blank line between entries
    irrecv.resume();              // Prepare for the next value
  }
  delay(200);
}

dham1012, 2017-06-29:
Doesn't seem to paste in very well

mdhiggins, 2017-06-29:
Wrap it in ``` ``` ``` at the beginning and end

dham1012, 2017-06-29:
'''#include <FS.h>                   // This needs to be first, or it all crashes and burns

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <DNSServer.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>          // https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <WiFiUdp.h>
#include <ESP8266mDNS.h>          // useful to access to ESP by hostname.local
#include <NTPClient.h>
#include <ArduinoJson.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>

#include <Ticker.h>               // For LED status

const int configpin = 13;         // GPIO13 (D7 on D1 Mini) to enable configuration (connect to ground)
const char *wifi_config_name = "IRBlaster Configuration";
int port = 80;
char passcode[40] = "ESP9951012";
char host_name[40] = "ESPIRremote";
String last_code = "";            // Stores last code
String last_code_2 = "";          // Stores 2nd to last code
String last_code_3 = "";          // Stores 3rd to last code
String last_code_short = "";            // Stores last code
String last_code_short_2 = "";          // Stores 2nd to last code
String last_code_short_3 = "";          // Stores 3rd to last code
String last_send_time = "";            // Stores last send code
String last_send_data = "";            // Stores last send code
String last_send_type = "";            // Stores last send code
String last_send_address = "";            // Stores last send code
String last_send_time_2 = "";            // Stores last send code
String last_send_data_2 = "";            // Stores last send code
String last_send_type_2 = "";            // Stores last send code
String last_send_address_2 = "";            // Stores last send code
String last_send_time_3 = "";            // Stores last send code
String last_send_data_3 = "";            // Stores last send code
String last_send_type_3 = "";            // Stores last send code
String last_send_address_3 = "";            // Stores last send code

String last_code_time = "";            // Stores last send code
String last_code_command = "";            // Stores last send code
String last_code_type = "";            // Stores last send code
String last_code_length = "";
String last_code_address = "";            // Stores last send code
String last_code_time_2 = "";            // Stores last send code
String last_code_command_2 = "";            // Stores last send code
String last_code_type_2 = "";            // Stores last send code
String last_code_length_2 = "";
String last_code_address_2 = "";            // Stores last send code
String last_code_time_3 = "";            // Stores last send code
String last_code_command_3 = "";            // Stores last send code
String last_code_type_3 = "";            // Stores last send code
String last_code_length_3 = "";
String last_code_address_3 = "";            // Stores last send code

ESP8266WebServer server(port);
HTTPClient http;
Ticker ticker;

WiFiUDP ntpUDP;

// You can specify the time server pool and the offset (in seconds, can be
// changed later with setTimeOffset() ). Additionaly you can specify the
// update interval (in milliseconds, can be changed using setUpdateInterval() ).
NTPClient timeClient(ntpUDP, "europe.pool.ntp.org", 3600, 60000);

bool shouldSaveConfig = false;    // Flag for saving data
[... 831 lines cut ...]
//    last_code_short = codeOutputSRT(&results);

    last_code_time_3 = last_code_time_2;
    last_code_command_3 = last_code_command_2;
    last_code_type_3 = last_code_type_2;
    last_code_length_3 = last_code_length_2;
    last_code_address_3 = last_code_address_2;
    
    last_code_time_2 = last_code_time;
    last_code_command_2 = last_code_command;
    last_code_type_2 = last_code_type;
    last_code_length_2 = last_code_length;    
    last_code_address_2 = last_code_address;
    
    last_code_time = codeOutputSRT(int(1), &results);
    last_code_command = codeOutputSRT(int(2), &results);
    last_code_type = codeOutputSRT(int(3), &results);
    last_code_length = codeOutputSRT(int(4), &results);
    last_code_address = codeOutputSRT(int(5), &results);    
  
    Serial.println("");           // Blank line between entries
    irrecv.resume();              // Prepare for the next value
  }
  delay(200);
}'''

dham1012, 2017-06-29:
I don't think it needs the bootstrap.min.js link, will remove tomorrow and test it without. Would be nice not to have to reference anything externally but I don't know how easy it is to include the css.

mdhiggins, 2017-06-30:
I removed the js and everything works fine
Really liked the look of this and went ahead and expanded on it to a 2 page system and added lots of features
Pushed it to the main branch, take a look, here's some screenshots

[image]
[image]

dham1012, 2017-06-30:
Fantastic, great additions. For me the time doesn't update (found this previously)

I've added 

'''  if (millis()>15012107 || !time_updated) {
    timeClient.update();
    time_updated == true;
  }'''
In the loop() routine. Seems to work, the ms might need increasing, or could be removed so it doesn't update ever after first run

mdhiggins, 2017-06-30:
Hm I'm not having the same issue with the time not updating
The timezone is UDP by default but otherwise its working fine for me without that line

mdhiggins, 2017-07-01:
Yeah tested it a few times today and time was working appropriately, albeit in UDP

Pushed some further updates after familiarizing myself with bootstrap more and trying to adhere to their standards, did some restructuring to make the code less as well

Ran the final HTML output through the W3 validation tool and passed for both pages, so I think the interface design is in a stable state

Thanks again for getting this started and introducing me to bootstrap, hadn't heard of it before but its great. The project looks nicer than most of the professional solutions out there these days

dham1012, 2017-07-01:
You've done a great job with it, looks great. Glad I could contribute in a small way.

dham1012, 2017-07-01:
This is where personal option comes in, but on the main page I do think it looks slightly clearer without the red within the send and received sections.  
One other point, on mobile devices, it doesn't seem to scale now, so it's all a bit large on the small screen, only reason this matters is it's quite useful when capturing codes to be able to use a phone rather than laptop.

mdhiggins, 2017-07-01:
I changed the mobile device initial scaling to 0.75 which helps get a little more onscreen but still has the UI change into mobile mode

dham1012, 2017-07-01:
That works much better, great, thanks.

I'm still getting intermittent results with getting the time on a restart. Seems hit and miss. I have added a delay just after it initiates, no idea if that will help

mdhiggins, 2017-07-01:
I pushed an update that changed around how the time works about 30 minutes ago, and made it so you can define your server/offset in the header

Try playing around with that and see if it helps

dham1012, 2017-07-01:
This was the version I have uploaded, I'll remove the delay and see how it performs without.

mdhiggins, 2017-07-02:
Updated the HTML output to use a continuous content stream instead of compiling everything into a string and then sending it all at once. Was running into occasional errors when the ESP would run out of memory on particularly long HTML strings
This way should be much more efficient and keep the memory free when generating the pages

dham1012, 2017-07-02:
Will give it a go, way beyond me now :) Great work.

I've linked this to Smartthings using webcore and ask alexa which integrates into the Echo. It works an absolute treat. 
I've created a wiki on what I've done, made references to your implementation, hope that is ok. I need to update some of the details regarding the smartthings bits as the developer of ask alexa is making some changes to make it all work better in this scenario.

http://thingsthataresmart.wiki/index.php?title=How_to_Control_your_TV_through_Alexa_and_Smartthings

I've also permanently wired the ESP, looks really neat, need to find a box to put it in.
[image]

mdhiggins, 2017-07-03:
Awesome write up, very thorough and excellent to see the smart things integration
And that permanently wired build is sexy. Any pictures from the underside? I'm gonna have to do that soon for some of mine

Would you have any objections to me linking to this wiki in the readme?

Also since you liked the project, I've got another one here

https://github.com/mdhiggins/RFSwitchController

Similar idea except this uses WeMo switch emulation and controls RF enabled outlets (or other RF devices) for up to 5 switches which Alexa recognizes as smart devices
RF receiver on GPIO 5 and transmitter on 4
Can set your on/off signals, pulse length, protocol, bit size
The web interface supports capturing codes to save them as well as manually typing them in or testing with a send code UI

When I get time I'm gonna do a readme for that now that the code is all working

dham1012, 2017-07-03:
More than happy for you to link yes, this is a great device you have put together and the more that can benefit and make use of various integrations the better.

Not much to see on the underside, bar my suspect soldering skills! It does look neat, and nice and compact.

[image]

Your other project sounds very interesting, I have some LightwaveRF devices which are integrated into Smarthings, but via their own hub, which in theory I could replace with this. Might have a look at when I'm finished with this, want to build another unit and put in lounge.
Not sure how easy it will be to capture the signals from these devices, its all a bit closed from what I understand (which is a little limited) Definitely worth some investigation.

mdhiggins, 2017-07-06:
Gonna go ahead and close this now, thanks for the all the input

closed 2017-07-06 by mdhiggins

mdhiggins, 2017-10-05:
Hey @dham1012  I have a new Amazon smart home skill I'm beta testing that allows directly creating smart home devices that work with the IR controller that won't require any coding or an amazon developer account. Trying to get some beta testers that I know are familiar with the project while awaiting approval from Amazon. If you're interested let me know and send me your Amazon account email and I'll send you some testing instructions