mdhiggins/ESP8266-HTTP-IR-Blaster #39: HTTP Server

goostoff, 2017-07-01:
The HTTP Server is responding very slowly in V3.  How can I collect additional information on this?

I'm not on my normal network - therefore it's possible it's something else, but it does seem odd.

mdhiggins, 2017-07-01:
One of the things it does on its first response is get the external IP address which requires pulling from an outside source (ddns) but it should cache that information after its first used for 72 hours

Is it slow on subsequent requests?

You can monitor the serial output of the device when performing HTTP commands for more information as to where the hang up is

goostoff, 2017-07-01:
What's the benefit to pulling the external IP address?  Can that be disabled if we don't need it?

goostoff, 2017-07-01:
I removed the code for pulling the external IP and live is much better.

mdhiggins, 2017-07-01:
Its used for the interface when capturing codes to generate the URL for you to use with IFTTT or other similar services

It shouldn't be too bad but I guess internet speeds and regional variability will impact this. The external IP is cached so once it has retrieved the external IP it should be instant for the next 72 hours

That being said I know some people may use this only locally and may not even have an external connection. I went ahead and added 2 new const bool options `getTime` and `getExternalIP` at the start of the blueprint. Changing these to false will disable retrieving the time/external IP

I also added some serial output that indicates how long the externalIP request took (averages about 100-200ms for me in testing, which isn't bad for a 72 hour request). 

Finally I made retrieving the externalIP part of the boot process (if enabled) so your first code transmitted or page request should be instant after the device is booted and may make the delay from retrieving less noticeable 

Hopefully these changes help

closed 2017-07-01 by mdhiggins