mdhiggins/ESP8266-HTTP-IR-Blaster #42: Resistors

mrk05s, 2017-07-06:
Great project! Thanks for taking the time to document and share this with the rest of us.

Electronics noob here, so please forgive what may seem to be a silly question, but which resistors do we need in place of R1 and R2 on the circuit diagram?

And will a 2N3904 work in place of a 2N2222? From what I can see they are identical except for the maximum current, with the 2N3904 rated at 200mA vs the 2N2222 at 1A.

mrk05s, 2017-07-06:
Just spotted issue #12, so ignore my first question.

mdhiggins, 2017-07-06:
2N3904 should work fine but depending on which LED you have you may want to delivery more current and the 2N2222 would be better. Make sure you're setting your resistors appropriately so you're not trying to pull more than the 200mA the 2N3904 is rated for

closed 2017-07-06 by mdhiggins