mdhiggins/ESP8266-HTTP-IR-Blaster #43: Send two digit code to Sony receiver

brainhunts, 2017-07-07:
I really love your project. I have been using it to remotely control an old Sony receiver in the garage.

However I run into a (probably) small issue: several of the buttons of my remote control, only generate two digit codes such as 16 for the first channel at the tuner. However when I send the code through http://.../msg?code=16:SONY:12&repeat=3&rdelay=20, nothing happens. Also using 016 doesn't work.

Any suggestions what to use?

Thanks in advance!

mdhiggins, 2017-07-08:
2 digit codes shouldn't be a problem or create any issues
Are larger/longer codes working without issue? also are you using the latest version of the software?

I assume you're using the repeat/redelay settings because its a SONY code, with the latest version this is no longer needed, it was fixed in the IRsend library and will now pulse the codes correctly

I would try updating and removing the repeat/rdelay and see if that helps at all

If that doesn't work you could also try to send the RAW command and see if that fixes your issue

brainhunts, 2017-07-09:
Thanks for your quick response.

I don’t know why, but the system now understands the two digit code indeed. Maybe it had something to do with the repeat command (need to check that further).

By the way, I wasn’t aware of the fact that a repeat wasn’t required any more for SONY devices, the readme still says something regarding Sony when it mentions the pulse code in the Json paragraph.

Thanks again for your help and the great work.

Van: Michael Higgins
Verzonden: zaterdag 8 juli 2017 07:39
Aan: mdhiggins/ESP8266-HTTP-IR-Blaster
CC: brainhunts; Author
Onderwerp: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Send two digit code to Sonyreceiver (#43)

2 digit codes shouldn't be a problem or create any issues
Are larger/longer codes working without issue? also are you using the latest version of the software?
I assume you're using the repeat/redelay settings because its a SONY code, with the latest version this is no longer needed, it was fixed in the IRsend library and will now pulse the codes correctly
I would try updating and removing the repeat/rdelay and see if that helps at all
If that doesn't work you could also try to send the RAW command and see if that fixes your issue
—

---
Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

mdhiggins, 2017-07-09:
Whoops look that one line got left behind after the update
I'll fix the readme

Thanks

closed 2017-07-09 by mdhiggins