mdhiggins/ESP8266-HTTP-IR-Blaster #50: PCB for this hardware

raptordemon, 2017-07-17:
Hi,

thanks for building this, just wanted to let you know that i built a PCB for this code.

its not been tested yet but i will report back when its been tested:
https://easyeda.com/raptordemon/ESP8266_IR_blaster-f8f224faba554bf3bb6ce0e32759fc23

cheers!
Raptor

mdhiggins, 2017-07-19:
This looks really cool
I don't have much experience with this, is the 1.3 revision using the latest pin assignments now?
I'd definitely like to order a couple of these if you've tried
I thought I had replied to this comment earlier and may be going crazy but don't see that reply now
Let me know how it goes! Is this using the V2 nodeMCU?

mdhiggins, 2017-07-19:
Never mind it was on the pull request that I commented, sanity restored

But yeah definitely keep me posted and send some pics when you get the PCB ordered, I'd like to check it out

raptordemon, 2017-07-19:
yea sorry about that, feel free to delete one. 

do these look correct?
int pinr1 = gpio14; // Receiving pin
int pins1 = gpio4; // Transmitting preset 1
int pins2 = gpio5; // Transmitting preset 2
int pins3 = gpio12; // Transmitting preset 3
int pins4 = gpio13; // Transmitting preset 4

its for the 25mm / 1in width node mcu like this one:
https://www.amazon.com/Generic-ESP8266-NodeMcu-Internet-Development/dp/B01N0B48NI/ref=sr_1_6?ie=UTF8&qid=1500467677&sr=8-6&keywords=nodemcu

im not sure if that counts as the v2

mdhiggins, 2017-07-20:
Yeah that's the V2, there's an unofficial V3 version that's larger that was made by some other company, that's why I asked

And those pinouts look correct yes

raptordemon, 2017-07-20:
great, i m going to reorder the boards this week and hopefully get it right this time.

mdhiggins, 2017-07-20:
One other question, it looks like the transmitters are using 3.5mm jacks? Maybe the receivers as well. What’s up with that?

Hasenpups, 2017-07-20:
Just as an information. 
I use the firmware with this hardware: https://www.openhardware.io/view/385/MySWeMosIRShield-IR-blaster-shield-for-WeMos-D1

mdhiggins, 2017-07-20:
That's pretty cool looking, but looks like no spot for a receiver?

mdhiggins, 2017-07-20:
Nevermind, just saw the receiver

Hasenpups, 2017-07-20:
There is a receive on one side between two LEDs. Someone already printed a housing.

https://forum.fhem.de/index.php/topic,72950.msg659524.html#msg659524

mdhiggins, 2017-09-30:
Looks like the GPIO15 reset option was causing people a lot of problems, I had to change to GPIO10

Any chance I could get you to at least update the small blueprint? I'm not familiar with this and afraid I'm gonna screw something up

raptordemon, 2017-10-02:
sure no problem

mdhiggins, 2017-10-04:
GPIO10 is working well for me on the NodeMCU board, can't speak to the other variants but I changed all my boards over to that without problems

Also, I have a new update hopefully coming soon and am looking for a better tester that uses the IR Blaster with Alexa (and I guess is in the US cause of Amazon's limitations). Any interest?

kokarn, 2017-10-04:
Just ordered all the parts, this looks amazing. I'll be using it with IFTTT & Google Assistant, is that any help?

PS: The PCBs looks great as well, will get those as well ^^

raptordemon, 2017-10-04:
HI pcb's updated, v1.5 is the latest it has gpio10 as the wifi reset. i have not tested this board as i am using the v1.3 and still have boards.

please let me know if it works
Raptor

mdhiggins, 2017-10-05:
Awesome thanks, I'll order some for testing

Also @raptordemon if you use your devices with Alexa, any interest in beta testing something?

raptordemon, 2017-10-05:
yw.

yea i can do that.

mdhiggins, 2017-10-05:
Send me an email with your amazon account email address and I'll add you as a beta tester for a new skill

ipjcrane, 2017-10-05:
It’s ipjcrane@hotmail.com<mailto:ipjcrane@hotmail.com>

mdhiggins, 2017-10-05:
Invite sent, and sent you an email explaining the new features

mdhiggins, 2017-10-18:
@ipjcrane @raptordemon 

Any luck with the beta testing?

If you need more help with setup here is a complete setup tutorial video that I finished making that will be live when Amazon enables the V3 API for all users
https://www.youtube.com/watch?v=JysKXOdrOlM

KDean7, 2017-10-24:
@raptordemon,
Are the circuit boards available yet?  Just starting this project and ordering parts now.  I have no breadboards and just heard about the samples you sent to mdhiggins.  Sounds like they'd be a much cleaner solution.

mdhiggins, 2017-10-25:
They are available. Use the v1.5 schematic when ordering. For first time builds though I would recommend using a breadboard for testing/debugging

w-brown, 2017-12-18:
Just wanted to say nice job!   I have built a prototype board using the ESP8266-12E module and am having fun playing with it.   I may decide to develop a board as well and share it out for those who want to reduce the size a bit as the NodeMCU is a bit bulky compared to the ESP module and a few surface mount components.

Do you have any plans to update the interface for the back end server?   One thing that would make it a bit more 'general user' friendly is if you had options not to have to enter the url for each command.  I feel if the interface simply asked for your ip, port, model of device, password, etc... and then for each command only required the hex codes for your remote it would make setting up various devices much simpler. 

In any case, great job!

mdhiggins, 2017-12-18:
Probably not
While it would make the interface a little cleaner it locks in less flexibility and it would also potentially break compatibility with all the people who have already defined their devices
In the current iteration you can actually combine controlling different controllers within the same virtual device
That being said it would be nice to reduce some of the redundant data. I'll have to see if I can come up with a clever solution that's backward compatible

zobotek, 2017-12-21:
@raptordemon 

0.  Nice job on the V1.5 PCBs!  Many thanks for sharing.

1.  I just ordered a batch of 10 large and 10 regular V1.5 boards from EasyEDA (China).  Got them and built them up today.  They were cheap - $26 got me 10 of each, including DHL shipping from China. Delivery was 3 days after ordering.  

@raptordemon & @mdhiggins 
2.  Only one problem:  configpin=GPIO10 crashed my NODEMCU board, even without being plugged in to the IR Blaster board.  Solution for me was to use configpin=16 (aka D0).  **It is NOT a good idea to use GPIO10 on a NODEMCU board because it's a reserved pin on some versions, including the NODEMCU Version 0.9 board which uses it for flash chip write control.**  0.9 NODEMCU crashes with pinmode(10,INPUT_PULLUP).  

@raptordemon
3.  Some suggestions for improvement:  
*  make the pin spacing for the resistors wider to accommodate 1/4W resistors
*  put down pads for a micro config switch in addition to a config jumper.  Putting a reset jumper in place on the smaller boards is inconvenient. 
*  put down pads for config mode on GPIO16 to support V0.9 NODEMCU (in addition to those on GPIO10 for V1.0)

mdhiggins, 2017-12-21:
I think at some point GPIO16 was the reset pin but was causing boot issues though I can’t remember exactly. It’s been quite frustrating finding a reliable reset pin but 10 hasn’t given me any trouble on the V2 nodeMCU which is the latest official

KDean7, 2018-03-09:
@raptordemon - This may sound really basic but I can't figure out how to order PCB's.  When I go to PCB Order on https://easyeda.com/order, it isn't clear on how to place an order.   Any suggestions?

katamara, 2018-03-17:
@KDean7 I was having a hard time too.  But I figured it out.  Here's what you need to do:

-go to: https://easyeda.com/raptordemon/ESP8266_IR_blaster-f8f224faba554bf3bb6ce0e32759fc23
-scroll down to the last board
-under the last board, there is a blue box that says "Open in Editor" 
-click that button and it will take you to a new page
-once it loads, there are tabs at the top for all the different board configurations
-choose the tab with the board you want
-next to the website logo , there is a drop down that lets you save the file
-choose save as (if you do not have an account, you will be prompted to create one)
-now choose Generate Gerber File
-at the bottom of the popup screen (you can order from here if you want) there is another button that says "Generate Gerber"
-now you have the gerber file that you can upload to the website that you want to order from.

Does any have any amazon or aliexpress links (or part#'s) for the wifi_reset button and/or the 3.5mm audio jack?

katamara, 2018-04-07:
> Does any have any amazon or aliexpress links (or part#'s) for the wifi_reset button and/or the 3.5mm audio jack?
I ended up ordering these jacks: https://www.aliexpress.com/item/20PCS-3-5mm-1-8-Female-Audio-Connector-5-Pin-DIP-Stereo-Headphone-Jack-PJ3024M/32689293903.html  they do fit but I have not been able to confirm that they work yet.  they do fit in the holes.  I will update once I can confirm they do indeed work.
for the reset button, I'm just using headers with a jumper.

>

katamara, 2018-04-07:
I ordered and received the following boards: 
IRBlaster v1.5 small
IRBlaster v1.5

I have soldered components on to the small board, but it's not transmitting IR.  The IR Receiver works fine.
I am wondering if I have 2n2222 and/or IR transmitter backwards.  Has anyone successfully used either of these boards?
If so, which way do these components fit in the board?  ie does the flat side of the component follow the outline on the board?  
Can you post a picture of the completed board?

Masonl23, 2018-04-28:
Hey , I got 1.5 board soldered everything up and The reciever works great but only the LED 1 to work and the others don’t. Btw i don’t have the 3 jacks soldered in, Without them does that mean it won’t finish the circuit and doesn’t power the other 3 LED’s? 
Thanks

wdavalil, 2018-08-20:
hi guys great work really, I order the boar 1.5 I solder everything in place except for the jacks that part I just confess I didn't understood and I still don't know how to do it im about to upload the code but not sure if everything is really in the place im kind of new in all this trying to learn. if someone have a pic showing how is the final result or similar would be really good. thanks ill be updating my progress.

raptordemon, 2018-10-11:
sorry for the delay, you should not need to populate the jacks for the extra leds to work. ill see if i can test mine

RienduPre, 2020-01-05:
> I ordered and received the following boards:
> IRBlaster v1.5 small
> IRBlaster v1.5
>
> I have soldered components on to the small board, but it's not transmitting IR. The IR Receiver works fine.
> I am [...]

Did you ever got it to work? I have the same issue?

TheMCMan, 2020-01-05:
Hey there.  Got mine to work correctly.  I have a schematic that shows how to correctly determine proper part orientation.  Will post here later today.

Do you have pic of your esp board to place here?  Not sure what you mean by a 1.5 board big and small..

RienduPre, 2020-01-05:
[image]
[image]

TheMCMan, 2020-01-05:
Hey there, The PCB you posted a pic of is interesting.  It looks like all of the parts are positioned correctly.  Like the flat side of the LED and Transistor are placed correctly, which means the polarity is correct.  I traced the lines from the parts the best I could and everything appears to be good.

Please also send a pic of the ESP8266 board you plug into the headers on the PCB.  Maybe your ESP8266 has a different pinout than what is needed by the PCB.

Attached is a schematic I developed and used for this project based on the project notes:
[image]
 
Here is the PCB i built using this schematic:
[image]

If you use plug connectors, for like the LED, you would be able to swap out the parts more easily to see if you had a part that was malfunctioning.  I am currently transitioning to these connectors for my builds.
[image]

I then put my PCB into a project box.
[image]

please send a pic of your ESP8266 board

RienduPre, 2020-01-06:
Here you go.
The only difference I see is that I use a 180 ohm resistor and you 10 ohm

[image]
[image]
[image]
[image]
[image]

TheMCMan, 2020-01-07:
That's the exact same ESP8266 board I use : )

Believe you nailed it with your keen observation there lol.  Going from a 10 ohm to an 180 ohm resistor would have a great impact in my opinion.  This resistor size directly controls the brightness of the Infrared LED.  

In the project video of IR Blaster Tutorial, at 1:32 minutes:seconds, mdhiggins references the resistance at 10 ohm as that would provide 500 mAmp current to the IR LED.

The IR Blaster Tutorial video is located here -- [https://youtu.be/JysKXOdrOlM](https://youtu.be/JysKXOdrOlM)

With a little experiment lol, maybe you can tell if you just have a rather dim IR LED bulb... find out exactly where the actual TV IR Receiver is located around the display screen.  What I normally do lol is put the remote like directly against the TV and move it around while pressing the on/off button.  When you identify that sweet spot that turns off the TV you can then hold your IRBlaster, with the IR LED pointed towards the TV of course, and send your commands away.

With any luck you will see that your device works but it's might be just a bit dim... then replacing the resistor will fix this.  

I did see many diagrams on Google images showing a higher value resistor like you used.  When building mine I tried to just base it off of the video.  But every hurdle in this project seems to be monumental lol.

Out of the 4 boards I've built, I couldn't get 1 to work..  never figured out why, but I think it had a lot to do with my despicable soldering..

RienduPre, 2020-01-08:
I can't get it to work. I gave up and look for other methods to operate my fire place

benlye, 2020-01-08:
Just buy one of these.  Works really nicely with Michael's code.
https://www.aliexpress.com/item/32958748327.html

See #236 for more information.

closed 2020-04-24 by mdhiggins

mdhiggins, 2021-08-25:
@raptordemon not sure if you're still using this project or not, but was wondering if you'd be interested in making a small update to your PCB. I continue to use the 1.5 small version to this day and its been awesome

I'm looking to tweak the design a little in a coming update, removing 4th IR led and subsequent hardware and replacing it with an input using a optocoupler.

The goal is to allow a 5v input/ground from a USB cable to be connected to the optocoupler and then to use GPIO13 as an input pin. This would allow me to detect the TV state using power from the debug USB port present on most TVs for reliable state monitoring to prevent repeat "on" commands from turning a TV off when discrete codes are absent.

I currently have the external/debug USB 5v port with a 220 ohm resistor and external ground connected to one side of the optocoupler with GPIO13 connected on the other side.

The mqttdev branch is updated to include the code

I tried modifying your PCB design myself but realized I have no idea what I'm doing.

You would be able to eliminate one of  the 2N222 transistors and one resistor and repurpose the other resistor for the optocoupler

Here is a quick photo of my test setup

[image]

Thanks

mdhiggins, 2021-09-13:
Built a few of the boards and its working great! Thanks for the update

raptordemon, 2021-09-13:
thats great news! thanks for the feedback!!