mdhiggins/ESP8266-HTTP-IR-Blaster #51: Need to change my setup - How do I reset the ino image with Arduino

feedrate, 2017-07-17:
When I upload the sketch it does not completely overwrite the previous setup - Is there a way to reset the image, or remove it before uploading anew?

_Could imagine that this is basic stuff, but as this is the first ESP8266 project I'm a bit at a loss of how to do this. And haven't been able to find a basic tutorial on this_

By the way thanks for this.
Works pretty great, been playing with a mosfet setup to remove the resistor for the gate. So fun stuff!

b0bje, 2017-07-17:
if you flash it with a dummy program (blinky led) it should clean up
everything aside from the data that was jsonified (which I believe should
only be wifimanager stuff).

If in doubt, flash the original lua firmware, should take of everything.

feedrate, 2017-07-17:
I did that with success a week ago. But that's not working now for some reason, so I thought that the previous success was a fluke.

It just starts up as normal:

> *WM: Adding parameter
> *WM: port_str
> *WM: 
> *WM: AutoConnect
> *WM: Connecting as wifi client...
> *WM: Using last saved values, should be faster
> *WM: Connection result: 
> *WM: 3
> *WM: IP Address:
> *WM: 10.0.0.8
> Default port changed
> WiFi connected! User chose hostname 'IR' passcode '12345' and port '3450'
> mDNS started. Hostname is set to IR.local
> URL to send commands: http://IR.local:3450
> HTTP Server started on port 3450
> External IP: 123.456.00.01
> External IP address request took 320 ms
> Ready to send and receive IR signals
> Turning off the LED to save power.

But did not know I needed to flash to get rid of all. Thanks for that tip. I'll look it up.

Will try again... 5min.

feedrate, 2017-07-17:
Tried with uploading the example WifiWebServer, and that worked better.
Able to change all. Cheers for that.

-Closed

closed 2017-07-17 by feedrate

mdhiggins, 2017-07-17:
Just out of curiosity what exactly did you need this for?

feedrate, 2017-07-18:
I changed my router, and with that, wanted to streamline/change all my
device names and passwords. With no ESP8266 experience (besides testing
this) I found my self stuck when uploading the sketch did not do the trick.

mdhiggins, 2017-07-18:
Oh to reset the wifi settings just connect D7 to ground and it will go into configuration mode.

feedrate, 2017-07-18:
Which you specified in the description (I just realised...).
Thank you! I'll give that a try next time.