mdhiggins/ESP8266-HTTP-IR-Blaster #52: Json "unknown" codes wont send. regular codes are fine.

moetop, 2017-07-22:
I am either missing something in reading the documentation, or there is an issue sending json. I take the code directly from the codes received, and paste it into a browser, but it doesn't seen to send the IR. Regular decoded IR codes (like for my samsung work), but unknowns don't send. Using latest code.

 Here are the html and serial output for a samsung and then the unknown.
http://192.168.200.130/msg?code=E0E040BF:SAMSUNG:32

http://192.168.200.130/json?plain=[{%27data%27:[600,1700,%20600,1700,%20600,1700,%20600,550,%20600,550,%20600,550,%20600,550,%20600,550,%20600,1700,%20650,1650,%20650,1650,%20650,550,%20600,600,%20600,550,%20600,550,%20600,550,%20600,550,%20600,1650,%20600,550,%20600,550,%20600,550,%20600,550,%20600,550,%20650,550,%20600,1700,%20600,550,%20650,1650,%20650,1650,%20650,1700,%20600,1650,%20600,1650,%20650,1650,%20650],%20%27type%27:%27raw%27,%20%27khz%27:38}]

Connection received - MSG
Blasting off
Blocking incoming IR signals
E0E040BF:samsung:32
Turning off the LED to save power.
Reenabling receiving

Connection received - JSON
Raw transmit
Blocking incoming IR signals
Sending code
Turning off the LED to save power.
Reenabling receiving

mdhiggins, 2017-07-22:
When you say they don't send, can you be more specific? It looks like they send based on the text output
I'm guessing the TV or whatever isn't turning on. Does the IR led blink? (use your phones front facing camera)
Also, can you post the non encoded URL without the escape characters?

red-box-labs, 2017-08-03:
I'm seeing a similar issue - I've checked the output with a scope

This one doesn't work - theres no output on the scope:
http://blaster.local:80/msg?code=3B1A11B5:UNKNOWN:34&pulse=2&repeat=5&pass=1234&out=1

This one works:
http://blaster.local/msg?code=5CB47:SONY:20&pulse=2&repeat=5&pass=1234&out=1

mdhiggins, 2017-08-03:
You can't send "unknown" codes like that, you have to send them as raw commands line he is doing above

ARRRDJ, 2017-08-05:
Having the same problem. Those codes aren't working.

mdhiggins, 2017-08-05:
No you can't send unknown codes
You can only send them in the RAW format

closed 2017-09-13 by mdhiggins