mdhiggins/ESP8266-HTTP-IR-Blaster #53: Hang when sending RAW

albertlt, 2017-07-23:
I decoded a RAW aircon signal using analysIR. I tried to send it as RAW but the device hang afterward. I have to press the reset button on the NodeMCU to make it work again. Below is the code I was trying to send:

http://192.168.1.2:80/json?plain=[{'data':[2900,-1800,405,-1125,405,-1125,405,-405,405,-405,405,-405,405,-1125,405,-405,405,-405,405,-1125,405,-1125,405,-405,405,-1125,405,-405,405,-405,405,-1125,405,-1125,405,-405,405,-1125,405,-1125,405,-405,405,-405,405,-1125,405,-405,405,-405,405,-1125,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-1125,405,-405,405,-405,405,-1125,405,-1125,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-1125,405,-1125,405,-405,405,-1125,405,-405,405,-405,405,-405,405,-405,405,-1125,405,-405,405,-1125,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-405,405,-1125,405,-405,405,-405,405,-405,405,-1125,405,-405,405,-405,405,-1125,405,-1125,405], 'type':'raw', 'khz':38}]

mdhiggins, 2017-07-23:
Looks like this was due to the negative numbers in your raw code at the odd positions
Odd positioned integers are essentially handled as delays and that data was being stored as an unsigned int
Negative unsigned int values get transformed into very large positive numbers so those negative numbers were creating a delay around 72 hours creating the hang

This is fixed now with the latest update

closed 2017-07-23 by mdhiggins

albertlt, 2017-07-23:
so is the negative number is actually a valid entry for the decoding process of your library or is there compatibility issue with the generated RAW format output of the AnalysIR software?