mdhiggins/ESP8266-HTTP-IR-Blaster #55: Sending codes in the format device:subdevice:OBC

DocEon, 2017-08-09:
Firstly, thanks so much for this! It's been a very fun project to work on.

I'm trying to make a blaster that'll work with an Onkyo receiver I have. I don't have the original remote, but I was able to find codes on http://irdb.tk/ that should work with my hardware. Unfortunately, I can't work out how to use any of these codes with this code. 

The simplest way of expressing the IR codes is the device:subdevice:OBC NEC format, but I can't figure out how to get from, say, 210:109:4 to the proper 'code' that I need to send my chip to actually send the right signal. IRDB also has raw stuff, but those commands have positive and negative signs and I can't figure out how to reconcile that with the example JSON provided.

Is there any way someone could provide some examples of taking an NEC IR code that we know the device, subdevice, and OBC information for and turning it into the right kind of URL to send to the chip?

I did try this: 
http://192.168.0.103/msg?code=210%20109%204:NEC:12&pulse=2&repeat=5&pass=mypass
The chip received the request and blasted ~something~, but I have no idea if it was the right code. It didn't turn on my receiver, in any case.

Thanks,

mdhiggins, 2017-08-09:
I would take this question over to 
https://github.com/markszabo/IRremoteESP8266

He writes the actual library that this program implements and would be better able to help you and the question is more regarding the library itself

DocEon, 2017-08-11:
Thanks for the response!

closed 2017-08-11 by DocEon