mdhiggins/ESP8266-HTTP-IR-Blaster #56: Config pin on wemos d1 mini

Johnwulp, 2017-08-18:
I have a wemos d1 mini. The config port specified in the ino file is 15, on my board, this is already ground. This is causing the software to always reset the wifi-manager.

mdhiggins, 2017-08-18:
Hm this is becoming a challenge due to the nature of what pins are used on the nodeMCU vs what pins are used on the wemos and balancing that with having 4 LEDs for output

For now I would just manually change the pins to what you want and I may end up just having a blueprint for wemos and nodemcu

Johnwulp, 2017-08-19:
You’re right. I changed it, and works great. It took me 30 minutes to figure out, that it was reset during restart. Luckily I could resolve it myself.

closed 2017-08-19 by Johnwulp