mdhiggins/ESP8266-HTTP-IR-Blaster #57: WifiManager still showing after connecting to network

shnax, 2017-08-19:
I configure Wifi through the IRBlaster Configuration access point. Then the device connects to my network. I can access it through its IP, but it displays the Wifi Manger instead of the main dashboard.

The led on the board is still blinking as though it is not connected, but it is.

[image]

pansters, 2017-08-19:
Same problem here

mdhiggins, 2017-08-19:
Make sure GPIO15 isn't connected to ground, that's the reset pin, it was recently changed from 13 to 15. If you're on wemos GPIO15 is always ground so I would change the reset pin to something different (or back to 13) in the blueprint

Conflicting with NodeMCU and WeMos

shnax, 2017-08-19:
Thanks! Changing the config pin back to 13 fixed the issue.
[image]

closed 2017-08-19 by shnax