mdhiggins/ESP8266-HTTP-IR-Blaster #60: 'USECPERTICK' was not declared in this scope

RaionHiro, 2017-09-02:
`/Users/test/Desktop/ESP8266-HTTP-IR-Blaster-master/IRController/IRController.ino: In function 'void codeJson(ArduinoJson::JsonObject&, decode_results*)':
IRController:809: error: 'USECPERTICK' was not declared in this scope
       r += results->rawbuf[i] * USECPERTICK;
                                 ^
/Users/test/Desktop/ESP8266-HTTP-IR-Blaster-master/IRController/IRController.ino: In function 'void dumpRaw(decode_results*)':
IRController:858: error: 'USECPERTICK' was not declared in this scope
     uint32_t x = results->rawbuf[i] * USECPERTICK;
                                       ^
/Users/test/Desktop/ESP8266-HTTP-IR-Blaster-master/IRController/IRController.ino: In function 'void dumpCode(decode_results*)':
IRController:891: error: 'USECPERTICK' was not declared in this scope
     Serial.print(results->rawbuf[i] * USECPERTICK, DEC);
                                       ^
exit status 1
'USECPERTICK' was not declared in this scope
`

I try to use MacOSX or Windows compiler upload, will have the same error, where is the problem?

argha123, 2017-09-05:
Hello Update / Downgrade  your IRremoteESP8266 Library Version 2.1.0 and then again compile...

mdhiggins, 2017-09-07:
Update coming shortly to fix compatibility with the new version

mdhiggins, 2017-09-13:
Fixed in latest update, should be compatible with latest version of IRRemote

closed 2017-09-13 by mdhiggins

consolacion, 2017-11-19:
same error on latest version (2.2.1) of IRRemoteESP8266  and version 2.2.3 of IRRemote

mdhiggins, 2017-11-19:
This has been fixed for a while now, compiles fine with 2.2.3, are you using the latest blueprint?

consolacion, 2017-11-19:
downloaded/installed it all today
IRRemoteESP8266  version 2.2.1
IRRemote version 2.2.3
OpenMQTTgateway-0.4

mdhiggins, 2017-11-19:
Should have IRRemoteESP8266 2.2.3 and might want to uninstall the regular IRRemote library as these can conflict and it is not required
Also OpenMQTTgateway is not used at all, not sure where you got there from, are you using the stock blueprint or a modified one?

And if you could post there error you're getting it would be helpful

consolacion, 2017-11-19:
IRRemoteESP8266 was the latest I could install via the manager, will try manual for 2.2.3
OpenMQTTgateway not used at all???? I use it, many people use it. That is how I got the error that led me here.

I am not sure what you mean by  "blueprint". I just installed the library

Error is:
/home/xxx/Arduino/OpenMQTTGateway-0.4/ZgatewayIR.ino: In function 'boolean IRtoMQTT()':
ZgatewayIR:66: error: 'USECPERTICK' was not declared in this scope
       rawCode = rawCode + (results.rawbuf[i] * USECPERTICK);

mdhiggins, 2017-11-19:
My mistake, your versions are correct IRRemoteESP8266 2.2.1 is the latest and IRRemote 2.2.3, I have both installed without issues

OpenMQTTgateway is not used at all. Its not part of this script, it doesn't matter if lots of people use it; if you're using a fork that implements that then you need to talk to whoever developed it. Your error is coming from a package that's not even part of this project. So I'm very confused as to why its even coming up unless you made modifications to the blueprint (the blueprint is the .ino file). You could try uninstalling that package for now but I am interested as to why that project is even coming up. Are you perhaps on the wrong github page?

consolacion, 2017-11-19:
Yes, as I said, a search for the error brought me here after a bit of clicking. wasnt  paying attention I got to a different project. apologies, i will  take it up at the openhabgateway  forum, again apologies