mdhiggins/ESP8266-HTTP-IR-Blaster #62: Getting type unknown on Samsung remote

nikki-reprap, 2017-09-16:
I am getting type unknown on a Samsung remote. Do i have to specify the protocol ?

Thanks

goostoff, 2017-09-23:
+1 - I'm having the same problem with Wifi with the latest code.

THEP0PE, 2017-09-26:
I am having the same problem as well. Also, after every time I enter my wifi information and allow the board to reboot, my wifi router's interface becomes unavailable. I have had to reboot the router every time in order to regain access to it. The wifi stays up and running, I just can't access the interface to view what devices are connected. Using Linksys EA6400.

mdhiggins, 2017-09-27:
Please see this already closed issue
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/58

closed 2017-09-27 by mdhiggins