mdhiggins/ESP8266-HTTP-IR-Blaster #63: NEC repeat code

vantees, 2017-09-23:
Hi.
Great work! I have IR bulb dimmer. To change brightness I need to send command hex code and then NEC repeat code three times or more. I am having rouble sending NEC repeat code. It shows as 0xFFFFFFFF (0bits not 32?!). Can anyone help me send NEC repeat command? 

Thanks

mdhiggins, 2017-09-27:
I'm not sure what you mean by NEC repeat command
Do you mean you need to send the command code multiple times?

vantees, 2017-09-29:
It is command that repeats last command send. The RAW code for it is: 9000,2250,560 and it is 38bit. When you send this code it will show as 0xFFFFFFFF and receiver will repeat the code received before it.

mdhiggins, 2017-09-30:
Ah I see
If its not decoding properly and you're getting what you see above you could try sending the command as a RAW command instead of encoded which may help

closed 2017-11-11 by mdhiggins