mdhiggins/ESP8266-HTTP-IR-Blaster #64: Wifi does not get saved. GPIO15 Not set to ground

simkhovich, 2017-09-27:
Every time i reboot it/loses power, i need to re configure the wifi setting.

mdhiggins, 2017-09-27:
Are you sure? That’s the only reason that would happen. What hardware you are using? NodeMCU? Did you try changing the reset configuration pin to something else and see if I makes a difference?

mdhiggins, 2017-09-28:
Did some digging and it looks like on certain NodeMCU builds GPIO15 defaults to the LOW state without being set to ground and this can cause and endless settings loop
Going to try changing the reset pin to GPIO10 (the quest for a good reset pin continues)
Just pushed an update today that changes the pin

mdhiggins, 2017-09-30:
Yeah looks like GPIO15 was no good, latest update uses 10

closed 2017-09-30 by mdhiggins

simkhovich, 2017-09-30:
Cool, thanks for your work.