mdhiggins/ESP8266-HTTP-IR-Blaster #65: compile error

wagnetor, 2017-10-04:
Hi, I get the following error:

Archiving built core (caching) in: C:\Users\...\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_c426d00b451c42e1e99a76ad49610f55.a
sketch\sketch_IRController.ino.cpp.o: In function `encoding(decode_results*)':

C:\D\ESP8266\Sketch\sketch_IRController/sketch_IRController.ino:888: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&)'

C:\D\ESP8266\Sketch\sketch_IRController/sketch_IRController.ino:888: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'

sketch\sketch_IRController.ino.cpp.o: In function `Uint64toString(unsigned long long, unsigned char)':

C:\D\ESP8266\Sketch\sketch_IRController/sketch_IRController.ino:888: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&)'

C:\D\ESP8266\Sketch\sketch_IRController/sketch_IRController.ino:888: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

Can everybody help me?

mdhiggins, 2017-10-05:
I would double check and make sure you're not missing of the top level imports on the blueprint, just download the blueprint and run the file directly and see if it compiles something is definitely missing from your install or blueprint

wagnetor, 2017-10-05:
http://arduino.esp8266.com/stable/package_esp8266com_index.json
here i must take version 2.3.0 - my old version 2.2.0 doesn't work
and
NTPClient-master
here i must take version 3.0.0 - my newer version 3.1.0 doesn't work

now i have no errors during the compilation 
Thx

closed 2017-10-05 by wagnetor

mdhiggins, 2017-10-05:
I'm able to compile with 3.1.0 without any issues

wagnetor, 2017-10-05:
Well, it wasn't the version number!
I had the false library

doesn't work:
https://github.com/stelgenhof/NTPClient

it works:
https://github.com/arduino-libraries/NTPClient

mdhiggins, 2017-10-05:
Ah gotcha. Strange but glad it’s all working