mdhiggins/ESP8266-HTTP-IR-Blaster #66: DENON doesn't work

toms001, 2017-10-07:
Hi there,

'DENON All Power Off' was decoded to
Command: 2A4C028C008E, Type: DENON, Length: 48, Address: 0X2A4C

A re-send to
Command: 2A4C028C008E, Type: DENON, Length: 48, Address: 0
does nothing.

I converted a ProntoHEX.xls-command
0000 006D 0000 0020 000A 001E 000A 0046 000A 001E 000A 001E 000A 001E 000A 001E 000A 0046 000A 001E 000A 001E 000A 001E 000A 0046 000A 0046 000A 0046 000A 001E 000A 001E 000A 0679 000A 001E 000A 0046 000A 001E 000A 001E 000A 001E 000A 0046 000A 001E 000A 0046 000A 0046 000A 0046 000A 001E 000A 001E 000A 001E 000A 0046 000A 0046 000A 0679
('DENON All Power Off')

to RAW
http://xxx.xxx.xxx.xxx/json?plain=[{'type':'raw','data':[263,789,263,1841,263,789,263,789,263,789,263,789,263,1841,263,789,263,789,263,789,263,1841,263,1841,263,1841,263,789,263,789,263,43572,263,789,263,1841,263,789,263,789,263,789,263,1841,263,789,263,1841,263,1841,263,1841,263,789,263,789,263,789,263,1841,263,1841,263,43572],'khz':38}]&pass=something

and the receiver switches off.

Is there any possibility to show the RAW-data of a decoded command?
I think this is a problem in the IRremote8266-library.

Thanks,
toms_01

mdhiggins, 2017-10-07:
The raw data should still be shown

toms001, 2017-10-07:
sorry, you are absolutely right, i must be blind... the following raw-code works too:

xxx.xxx.xxx.xxx/json?plain=[{'type':'raw','data':[3302,1764,436,404,434,406,436,1242,434,406,434,1242,438,402,436,1220,458,402,436,402,438,1240,436,382,456,406,436,1220,454,1224,458,404,406,434,436,402,438,380,458,382,454,384,458,404,436,404,436,1240,436,382,458,1242,436,404,436,402,434,384,456,1244,436,1220,432,432,438,402,432,384,458,404,434,384,456,404,434,384,428,434,408,432,434,384,456,1222,456,406,434,384,428,410,430,1272,408,1270,406,1272,408,430,408],'khz':38}]&pass=something
according to IrScrutinizer it is protocol = Denon-K, device = 4, subdevice = 1, obc = 3
(above protocol = Denon, device = 2, obc = 226)

mdhiggins, 2017-11-11:
I would raise this issue over here
https://github.com/markszabo/IRremoteESP8266

since its inherently an issue with the IR remote library

closed 2017-11-11 by mdhiggins