mdhiggins/ESP8266-HTTP-IR-Blaster #68: control with openhab

saikhurana, 2017-10-15:
hey i just wanted to as if anyone has ever integrated this with openhab using http binding. i was able to control this through webhook but not directly on the local server. the openhab spits out some authentication error 
HTTP/1.1 401 Unauthorized
i want to control it directly with my openhab server. 
also i want to request the writer of this code to make this with mqtt protocall.

mdhiggins, 2017-10-15:
No plans to do MQTT or further integration with other service, the large complex multistep commands don’t play nice with MQTT since they contain a lot of data. Also the ESP8266 has authentication issues with AWS IOT devices making this project not ideal for that

There’s going to be a new Alexa skill that adds native smart home commands to the device though using the V3 version of their API. Its already approved and will be published once Amazon opens up V3 to consumers at the end of the month, letting you control your devices using natural language and not depend on IFTTT/Trigger

mdhiggins, 2017-10-16:
If you're interested in beta testing the new skill prior to release let me know

saikhurana, 2017-10-16:
i really wish i could beta test this sketch but i have no knowledge about amazon echo skills(the coding part). if you can teach me i can beta test :-)

mdhiggins, 2017-10-16:
You actually need no coding knowledge at all, that's the beauty of the new skill!

Email me your amazon account email address and I'll add you as a beta tester (mdhiggins23@gmail.com)

saikhurana, 2017-10-17:
Done!!Check Your Inbox

closed 2017-11-11 by mdhiggins