mdhiggins/ESP8266-HTTP-IR-Blaster #69: URL from raspberry pi

wagnetor, 2017-10-16:
Hello,
i have a problem with the URL request from the raspberry pi.
With safari on my ipad it works like a charme, but how can i request the URL from my wemo enabled raspberry pi. I need a request from bash for the ActionURL.

#!/bin/bash
sudo wget http://esp.local/msg?code=20DF10EF:NEC:32

doesn't work!
It's only download the html-webside

has everybody an idea for me?

mdhiggins, 2017-10-16:
You can add ‘&simple=1’ to your url to stop the html output and get minimal plain text. 

http://esp.local/msg?code=20DF10EF:NEC:32&simple=1

wagnetor, 2017-10-18:
with curl and '  ' it works fine!

#!/bin/bash
sudo curl 'http://esp.local/msg?code=20DF10EF:NEC:32'

closed 2017-10-18 by wagnetor