mdhiggins/ESP8266-HTTP-IR-Blaster #71: Alexa Skill?

primexa, 2017-11-01:
Hello,
I have done all the steps required for this project and was hoping for an update on the Alexa Skill. Amazon Alexa app updated a couple days ago and I was hoping to see the IR Controller skill on the Alexa Skills store but does not seem to be active yet as of November 1st :(  Do you have any update as I am sure you are anxious as well?

mdhiggins, 2017-11-01:
It’s finished and approved by amazon. Just waiting for the official push for all the V3 smart home skills. They told me end of October but haven’t given a specific date and obviously that has passed. It’s all in their hands now.

primexa, 2017-11-01:
Thank you for the quick response, waiting patiently....

mdhiggins, 2017-11-01:
Did I sent you a beta invite already?

primexa, 2017-11-01:
No sir, I have gotten a beta invite. That would be fantastic!

primexa, 2017-11-01:
Sorry, let me reiterate, no I have NOT gotten an invite :)

mdhiggins, 2017-11-01:
Send me your Amazon account email address

primexa, 2017-11-01:
sent...

mdhiggins, 2017-11-02:
Invite sent

Pawand15, 2017-11-02:
Hi could you please send me an invite for the skill
Amazon account details: pawan.d@outlook.com

Really a great project.. Already made the hardware!
Just received my echo In India.

kickaxe60, 2017-11-11:
This works for me.  How do we add channel changing capabilities?

Thanks

mdhiggins, 2017-11-11:
It’s in development for the next update

kickaxe60, 2017-11-11:
I would be happy to be a beta tester if you need one.  Note that I couldn't get the Input Controller working, no matter what label I gave the input.  I could switch inputs by tricking Alexa by assigning the IR code for input switching using a Playback Controller instead though.

mdhiggins, 2017-11-11:
That’s probably more of an issue on Alexa’s end. I too had trouble getting that to work at first. It’s only very specific input names and it can be picky about the device name. I use it to change my speaker system around and the following commands work for me:

Alexa change surround sound input to Bluetooth 
Alexa change surround sound input to TV

If it’s still not working I could recommend using the scene controller to make a scene that changes the input. That scene can then be used with a routine to let you say whatever you want.

kickaxe60, 2017-11-11:
OK I'll try some variations of what you suggest for input switching.  Do you have any info on how to create a scene.  Can you have more that one or can one scene have many triggers?

Thanks

mdhiggins, 2017-11-11:
You can create many scenes. The scene controller setup options are at the bottom of  the page (tehpsyc.pythonanywhere.com)

From: kickaxe60
Sent: Saturday, November 11, 2017 7:15 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

OK I'll try some variations of what you suggest for input switching. Do you have any info on how to create a scene. Can you have more that one or can one scene have many triggers?
Thanks

kickaxe60, 2017-11-11:
I can't seem to figure out how to use scenes.  I can enable and put command activate/deactivate commands but that's it.  I click reference but can't see any information on how to use scenes...

mdhiggins, 2017-11-11:
Just add the URL like you would any other command, update and save, then rerun device discovery

Now you can say “turn on SCENENAME” and “turn off SCENENAME”, or saying “activate SCENENAME” or “deactivate SCENENAME”

Scenes can then also be incorporated into routines

From: kickaxe60
Sent: Saturday, November 11, 2017 7:40 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

I can't seem to figure out how to use scenes. I can enable and put command activate/deactivate commands but that's it. I click reference but can't see any information on how to use scenes...

kickaxe60, 2017-11-11:
So can you can have just one scene per device, and all other controller info e.g. power/on off, step volume... are ignored for that device.  Is that correct? If not, how do I add more than one scene for a particular device? Basically, I would like to add commands like "Alexa, switch TV to CNN"...  I'm just trying to wrap my head around how scenes work.  

Thanks for your help Michael, it's appreciated.

Denis

mdhiggins, 2017-11-11:
Treat the scene as its own device. You can still set up all the other functionality just do it separately. (The URLs can point to the same ESP8266, that’s fine). 
So you would have your TV with its power and volume control as one device, then just create a scene and name it HDMI1, then another scene and name it Bluray, etc

For things like changing channel to CNN, that is something that I’m working on support for via the channelcontroller but is still in development, so it should be natively supported soon

From: kickaxe60
Sent: Saturday, November 11, 2017 7:50 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

So can you can have just one scene per device, and all other controller info e.g. power/on off, step volume... are ignored for that device. Is that correct? If not, how do I add more than one scene for a particular device? Basically, I would like to add commands like "Alexa, switch TV to CNN"... I'm just trying to wrap my head around how scenes work.
Thanks for your help Michael, it's appreciated.
Denis

kickaxe60, 2017-11-11:
Got it!  Is there a limit as to how many devices Alexa will discover i.e. can I have 20 scenes?

mdhiggins, 2017-11-11:
I’m not sure, I don’t think there is, there’s a character limit for device discover packet size but its crazy high and you probably won’t hit it without hundreds of items

From: kickaxe60
Sent: Saturday, November 11, 2017 7:54 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

Got it! Is there a limit as to how many devices Alexa will discover i.e. can I have 20 scenes?

kickaxe60, 2017-11-11:
OK Thanks Michael.

kickaxe60, 2017-11-14:
Hi Michael, I added many scenes yesterday and all work fine.  Today, I want
to add a few more and discovery doesn't find anything and all the other
scenes say offline, although they still work.  Any idea what would cause
that and how I woud go about getting Alexa to discover my new scenes
created through IR Controller?

Thanks

Denis

mdhiggins, 2017-11-14:
I pushed some security updates and appear to have created a bug not revealed with my initial testing. At work but will fix or roll back those changes tonight

mdhiggins, 2017-11-15:
Should be resolved now. Just an issue with one of the security tokens

kickaxe60, 2017-11-15:
Hi Michael, still no luck here. I tried disabling and re-enabling the IR
Controller skill, sign out and back in again and they still show offline.
I have a smart plug which is discovered no problem but none of the IR
Controller virtual devices or scenes get picked up.  I'm away for a few
weeks on holiday and will try again when I get back.

Thanks for your help.

Denis

kickaxe60, 2017-12-07:
Hi Michael, back from holiday and trying this out again.  I see you've
updated the .ino file and added channel capabilities.  I seem to have
everything setup correctly however, when I try to send a code I get the
following in terminal:

Connection received - MSG
Failed security check, signature is too old
Server: 0
Local: 1512667631
MID:

I'm using the amzn1 account at the bottom of the controller setup.  Any
ideas what my issue might be?

Thanks

Denis

mdhiggins, 2017-12-07:
Looks like it’s unable to connect to the time server to verify your time stamp. This is needed for authentication. 

Make sure you have all that stuff configured and have your DNS settings configured appropriately. That caused me a major headache before with the same issue.

mdhiggins, 2017-12-07:
Wait looking at that closer it looks like that’s not the case. The local time stamp worked but the server did not. Are you sending the command from a regular URL or from Alexa? Looks like the server time stamp was not included? Once the user_id is set the device goes into Alexa only mode and won’t accept regular URLs anymore.

kickaxe60, 2017-12-07:
Thanks, got the basic stuff working again.  I am not able to get channels
to work though.  My echo dot says "this device doesn't support that."  I
have to use raw data as my cable box remote is not recognized.  I can get
them working with scenes by firing off the three numerical codes but that
means creating many channel scenes...
Here's my base url with ip x'd out:
http://xxx.xxx.xxx.x:667/json?pass=pass
Here's a sample of the button 1 syntax:
'data':[8996,4454,514,4454,516,2220,514,2190,544,2218,516,2188,538,2226,514,2190,546,2186,542,2222,510,2218,514,2222,514,2190,546,4454,512,4428,546,4426,546,4424,544],'type':'raw’,’length’:35

Could it be a syntax thing where I code the channels?

mdhiggins, 2017-12-07:
Looks like your issue is stemming from the fact that you’re using 2 different characters for your single quote and its confusing the JSON object

You have ' and ’

I pasted yours and mine also malfunctioned

If I use the following it works fine, just use the first character for your quote:

data:[8996,4454,514,4454,516,2220,514,2190,544,2218,516,2188,538,2226,514,2190,546,2186,542,2222,510,2218,514,2222,514,2190,546,4454,512,4428,546,4426,546,4424,544],type:'raw'

You don’t need single quotes around the key (data/type) and length does nothing for raw codes

Also you should really try using the encoded version if possible instead of RAW, what kind of remote/TV?

From: kickaxe60
Sent: Thursday, December 7, 2017 4:47 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

Thanks, got the basic stuff working again. I am not able to get channels
to work though. My echo dot says "this device doesn't support that." I
have to use raw data as my cable box remote is not recognized. I can get
them working with scenes by firing off the three numerical codes but that
means creating many channel scenes...
Here's my base url with ip x'd out:
http://xxx.xxx.xxx.x:667/json?pass=pass
Here's a sample of the button 1 syntax:
'data':[8996,4454,514,4454,516,2220,514,2190,544,2218,516,2188,538,2226,514,2190,546,2186,542,2222,510,2218,514,2222,514,2190,546,4454,512,4428,546,4426,546,4424,544],'type':'raw’,’length’:35

Could it be a syntax thing where I code the channels?

kickaxe60, 2017-12-07:
I'm with Cogeco cable in Ontario.  With their set top box they use an Atlas
DVR 5-device universal remote.

I pasted your data and still no joy.  Did my base url structure look OK? I
did not get a not supported message though, the dot just flicked around
blue for a few seconds with no feedback.
How are you structuring your verbal command?

I'll keep plugging away.  Thanks again for your help.

kickaxe60, 2017-12-07:
Also, perhaps my channel up/down syntax is messing things up as this is not
working either... This is the raw syntax I'm using for channel up:
http://xxx.xxx.xxx.x:667/json?pass=pass&plain=[{'type':'raw
','data':[8850,4450,500,4450,450,4400,550,2200,500,4400,500,2200,500,2200,500,2200,550,2150,550,2200,500,2200,500,2200,500,2250,450,4400,550,2200,500,4400,500,2200,500],'khz':38}]

I see now that we must supply a number to increment by for channel up so I
believe my syntax is missing something.

mdhiggins, 2017-12-08:
Gonna need you to be more specific

The channel increment using JSON you will need to specify a repeat value somewhere, should look like this:

http://xxx.xxx.xxx.x:667/json?pass=pass&plain=[{'type':'raw
','data':[8850,4450,500,4450,450,4400,550,2200,500,4400,500,2200,500,2200,500,2200,550,2150,550,2200,500,2200,500,2200,500,2250,450,4400,550,2200,500,4400,500,2200,500],'khz':38,'repeat':*chan*}]

Also if you could email me your user_id that will be helpful for me to pull the logs and further evaluate

If you say “change TV channel to 50” for example, what response do you get? Does the light come on? If you check your device send history do you see the commands being sent? If the device lights up that’s a good sign it means the command was at least processed

From: kickaxe60
Sent: Thursday, December 7, 2017 6:43 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

Also, perhaps my channel up/down syntax is messing things up as this is not
working either... This is the raw syntax I'm using for channel up:
http://xxx.xxx.xxx.x:667/json?pass=pass&plain=[{'type':'raw
','data':[8850,4450,500,4450,450,4400,550,2200,500,4400,500,2200,500,2200,500,2200,550,2150,550,2200,500,2200,500,2200,500,2250,450,4400,550,2200,500,4400,500,2200,500],'khz':38}]

I see now that we must supply a number to increment by for channel up so I
believe my syntax is missing something.

kickaxe60, 2017-12-08:
I tried adding the ,'repeat':*chan* to the channel up syntax and that did
not work.  When I tell Alexa to “change TV channel to 50” it now says this
device does not support that.  A while ago, it would just blink and say
nothing.  I've sent my amzn1 user id to your email.

mdhiggins, 2017-12-08:
So interestingly after reviewing the logs it doesn’t look like any of your requests are actually making it to the skill
I see lots of activity on your account but none for adjusting or changing the channel

Looking through your discovery packet it looks like everything should be good to go, but whatever way you’re phrasing things Alexa isn’t even processing the commands to send it to the skill

The official phrasing for channel change is 

“Alexa, change channel to 200 on device”

And the official phrasing for the channel adjustment is

“Alexa, channel up on device”

Try speaking in that format

Also, I notice you’re using the scene controller for input changing instead of the inputController
Were you having issues with that as well? I see HDMI1/HDMI3 

If you set  the input controller and name your inputs hdmi 1 (with a space) that should work natively

From: kickaxe60
Sent: Thursday, December 7, 2017 7:51 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

I tried adding the ,'repeat':*chan* to the channel up syntax and that did
not work. When I tell Alexa to “change TV channel to 50” it now says this
device does not support that. A while ago, it would just blink and say
nothing. I've sent my amzn1 user id to your email.

kickaxe60, 2017-12-08:
OK Thanks for spending the time Michael.  I thought things were set up
properly but for some reason, Alexa keeps saying the device does not
support that.  Even the inputs just do not want to work with the TV
device.  I'm pretty sure I tried naming them with and without a space but
I'll try that again to be sure.  I can get channels and inputs to work no
problem using scenes but that is really a cumbersome way to accomplish what
I'm trying to do so at this point, I'm at a loss.  I did try the official
phrasings from the references listed in the controller setup but still no
luck. I'll keep troubleshooting when I get the time.  If you can think of
something else to try, please let me know. I appreciate your help.

mdhiggins, 2017-12-08:
Keep me posted, I’ll try to think of some other solutions
The other weird thing you can do is try a different name for your device itself, I’ve head Alexa get intermittently confused with just plain ‘TV’ sometimes
The Amazon devs are actively revising and attempting to improve the new API and I’ve seen them break and restore various things during development, they’re still making changes on a daily basis so hopefully things will improve

The other thing to check is in your Alexa history to see exactly what phrasing Alexa is hearing when you say your commands that don’t get sent (IE that device does not support that)

From: kickaxe60
Sent: Thursday, December 7, 2017 10:09 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

OK Thanks for spending the time Michael. I thought things were set up
properly but for some reason, Alexa keeps saying the device does not
support that. Even the inputs just do not want to work with the TV
device. I'm pretty sure I tried naming them with and without a space but
I'll try that again to be sure. I can get channels and inputs to work no
problem using scenes but that is really a cumbersome way to accomplish what
I'm trying to do so at this point, I'm at a loss. I did try the official
phrasings from the references listed in the controller setup but still no
luck. I'll keep troubleshooting when I get the time. If you can think of
something else to try, please let me know. I appreciate your help.

mdhiggins, 2017-12-08:
I’m actually surprised the kHz thing made a difference, when kHz is not specified it defaults to 38 so unless you needed a different frequency then that shouldn’t have caused issues. But either way I’m glad its work

The reference example for HDMI is wrong on Amazon’s part and I hadn’t realized it either until yesterday. I updated the site to regex check for HDMI# input names and automatically insert the space where appropriate so future users shouldn’t run into that issue

It may be a while for the channel names to work; I’ll need a good free API that contains TV listings to be searchable to link a network to a channel and so far all the ones I can find are paid and require monthly subscription fees from the developer which is out of my price range for this free project. I’ll keep trying to look up and think of alternatives

Glad everything is working for you though!

From: kickaxe60
Sent: Friday, December 8, 2017 8:51 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

Hi Michael, well after a good nights sleep, and a bit of reading on your
GitHub page this morning, I noticed that raw requires kHz data. Once I put
those in the channel numbers, they worked fine. I also spaced out HDMI and
1 or 2... and that too worked in the TV input section. The reference
example showed HDMI1 (which is what I used) rather than HDMI 1, so adding
the space fixed it.

Now I'm looking forward to selecting channels by station/network name when
that becomes available.

Thanks again for your help.

Denis

kickaxe60, 2017-12-08:
I had a separately named device named COGECO, which is for my cable box,
that I used for the channels and decided to incorporate the channels in my
TV device instead, even though they use different remotes so maybe that had
something to do with it i.e. Alexa likes TV commands and not COGECO when it
comes to channels...

I can easily create scenes for station names for the few that I watch
regularly so that will work in the meantime.  Another cool option would be
to implement guide functions within the device but that too I can use
scenes for to navigate...

Thanks again.

mdhiggins, 2017-12-08:
Well the ChannelController API supports network names, so saying something like “change channel to ABC” should work
The caveat is that the backend code needs some way to translate ABC into a numerical channel on a per user basis

One option I’m considering is rather than depending on a specific API is to allow users to create their own channel map and have it just contact the pythonanywhere server for that data and keep on the user to create a map of networks they care about

I’ll have to play around with it and try and find the best solution

From: kickaxe60
Sent: Friday, December 8, 2017 10:06 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

I had a separately named device named COGECO, which is for my cable box,
that I used for the channels and decided to incorporate the channels in my
TV device instead, even though they use different remotes so maybe that had
something to do with it i.e. Alexa likes TV commands and not COGECO when it
comes to channels...

I can easily create scenes for station names for the few that I watch
regularly so that will work in the meantime. Another cool option would be
to implement guide functions within the device but that too I can use
scenes for to navigate...

Thanks again.

kickaxe60, 2017-12-08:
That's kind of what I'm doing with scenes.  I'm calling each one by channel
name e.g. TSN1 and firing off the numerical codes in sequence to tune into
that channel.  So far it seems to work fine. It's a bit of a pain but you
only have to do each one once as the channel numbers rarely change.

mdhiggins, 2017-12-08:
Just pushed a big update to the setup site
You can now add more than 4 input options via the input controller
But more importantly.. I added support for channelController stations
You can define your stations and their channel numbers now and then verbally change channels based on those names; please give it a try and let me know if its working

Mike 

From: kickaxe60
Sent: Friday, December 8, 2017 10:54 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

That's kind of what I'm doing with scenes. I'm calling each one by channel
name e.g. TSN1 and firing off the numerical codes in sequence to tune into
that channel. So far it seems to work fine. It's a bit of a pain but you
only have to do each one once as the channel numbers rarely change.

kickaxe60, 2017-12-08:
Will check it out tomorrow. 
Thanks

kickaxe60, 2017-12-09:
> ESP8266-HTTP-IR-Blaster/issues
> auth/AgAYdesC9
> auth/AgAYdYFIBegTMe2aAZ_

When I click on Define your channels I get:
Error!
There was an error processing your saved JSON object, it is likely
corrupted or invalid and will need to be recreated. Scroll to the bottom of
this page and try to salvage it.

I'm a little hesitant to try this until I'm sure I won't lose what I've
done so far...

mdhiggins, 2017-12-09:
Whoops, that error is an accident, should be the welcome message
Fixed now

Additionally, you can backup your whole config but copying the JSON object to a text file on the main page at the bottom, if ever there’s a problem pasting that back in that location will restore all your devices and settings

Let me know how it goes
Mike

From: kickaxe60
Sent: Friday, December 8, 2017 7:33 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

When I click on Define your channels I get:
Error!
There was an error processing your saved JSON object, it is likely
corrupted or invalid and will need to be recreated. Scroll to the bottom of
this page and try to salvage it.

I'm a little hesitant to try this until I'm sure I won't lose what I've
done so far...

kickaxe60, 2017-12-09:
OK I have made a backup of my JSON file.  I will play with this tomorrow
morning and let you know how I make out. Were there changes to the ino
sketch or just the controller setup?

mdhiggins, 2017-12-09:
Just the setup and Alexa backend. No ino changes

kickaxe60, 2017-12-09:
Also, any particular way the voice command needs to be structured for Alexa?

mdhiggins, 2017-12-09:
Nah it’s structured pretty much as you’d expect. I’m gonna have to add some extra logic to try and reduce errors depending on how Alexa formats the channel name but not uncomplicated common stations it’s working well.

kickaxe60, 2017-12-09:
> pass=pass&plain=[{'type':'raw
> auth/AgAYdb-
> auth/AgAYdd-
> auth/AgAYdVetwdjVqWc-

When I add a station name and number and go to update channels, I get the
same corruption error...

mdhiggins, 2017-12-09:
Try now. Should be fixed.

kickaxe60, 2017-12-09:
Hi Michael, it's very hit and miss at this point.  I set up 14 of them and
only three work as expected. The first one is called cogeco on 700 and it
recognizes cogeco but only does a channel up.  The next two, ctv and cbc
are not recognized (Alexa responds: that command doesn't work on device
tv). I created a scene called cbc for channel 702 and it works flawlessly.
The next two, city and global work as expected.  The next  8 channels,
chch, omni1, omni2, abc, cbs, nbc, pbs, and fox, get the same error from
Alexa. The last one called tsn 1 works as expected.  I tried with and
without spaces on the lettered networks and that made no difference.  The
command that I use is: Alexa, change TV channel to global. I've tried a few
variations for those that don't work with no luck...

mdhiggins, 2017-12-09:
Alright I updated my algorithm for matching channels to be more robust with ways Alexa may interpret voice names for channels
I actually copied your channel data and tested them all now and they all appear to be recognized

Cogeco is the only one that is iffy, you have to pronounce it with a ‘j’ sound cojecko without an accent on the final ‘co’ and then it recognizes it

Give it a shot

From: kickaxe60
Sent: Saturday, December 9, 2017 8:39 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

Hi Michael, it's very hit and miss at this point. I set up 14 of them and
only three work as expected. The first one is called cogeco on 700 and it
recognizes cogeco but only does a channel up. The next two, ctv and cbc
are not recognized (Alexa responds: that command doesn't work on device
tv). I created a scene called cbc for channel 702 and it works flawlessly.
The next two, city and global work as expected. The next 8 channels,
chch, omni1, omni2, abc, cbs, nbc, pbs, and fox, get the same error from
Alexa. The last one called tsn 1 works as expected. I tried with and
without spaces on the lettered networks and that made no difference. The
command that I use is: Alexa, change TV channel to global. I've tried a few
variations for those that don't work with no luck...

kickaxe60, 2017-12-09:
Still have issues with ctv, chch, and cogeco.  Omni1 & 2 work sometimes.
I'll keep plugging away.  Is there a limit I should be worried about?

mdhiggins, 2017-12-09:
No limit to worry about

I notice you changed your channel names and added spaces, the best formatting is one with no spacing between letters (ABC, NBC etc). The algorithm strips out spaces and punctuation created by Alexa to try to match with something in your list

Using the following all of these work except cogeco just cause Alexa interprets things weird for that one, might need to contact Alexa since that’s apparently a big Canadian network they should recognize

[{
		"name": "cogeco",
		"channel": "700"
	}, {
		"name": "ctv",
		"channel": "701"
	}, {
		"name": "cbc",
		"channel": "702"
	}, {
		"name": "city",
		"channel": "704"
	}, {
		"name": "global",
		"channel": "706"
	},
	{
		"name": "chch",
		"channel": "707"
	}, {
		"name": "omni 1",
		"channel": "708"
	}, {
		"name": "omni 2",
		"channel": "709"
	}, {
		"name": "abc",
		"channel": "712"
	}, {
		"name": "cbs",
		"channel": "713"
	}, {
		"name": "nbc",
		"channel": "714"
	}, {
		"name": "pbs",
		"channel": "715"
	}, {
		"name": "fox",
		"channel": "716"
	}
]

From: kickaxe60
Sent: Saturday, December 9, 2017 10:46 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; State change
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Alexa Skill? (#71)

Still have issues with ctv, chch, and cogeco. Omni1 & 2 work sometimes.
I'll keep plugging away. Is there a limit I should be worried about?

kickaxe60, 2017-12-09:
OK, I'll remove the spaces. I'll maybe change the cogeco channel to
something else. I've added a bunch more but have not tested them.  Won't
likely get to that until tomorrow and will let you know how I make out.

Great work on this by the way.  My wife loves not having to fiddle with
remotes.

kickaxe60, 2017-12-10:
Hi Michael, I pretty much have all of the station names working now.  It
would be nice to have some guide functionality built in too e.g. turn on
guide, show favourites, page up/down, scroll up/down, show info, select,
last channel, exit...

I also have a bit of an issue with one of the PVR commands.  I can say
Alexa, Rewind/Fast Forward/Pause/Stop PVR but Play PVR results in Alexa
looking to play music by a band named PVR in amazon music.  If I say Alexa,
PLAY ON the PVR it works but it's a bit awkward.

Thanks for your help.  Keep me posted on any other enhancements, I'd be
glad to test them out for you.

Take care,

Denis

mdhiggins, 2017-12-10:
There’s no guide functionality built into the API so for now that’s not possible. Will require amazon support to expand that. 

The play issue is known and is another Alexa issue unrelated to my skill. They seem to be actively tweaking their algorithm to improve this so hopefully it will get better. 

Glad the channel controller is working better now though!

kickaxe60, 2017-12-10:
Not a biggie re: the guide as I have the eight or so commands set up as
scenes and they work fine.

[18 more replies]