mdhiggins/ESP8266-HTTP-IR-Blaster #74: error connecting to wifi

pranaystark, 2017-11-16:
*WM: OnePlus 5
*WM: -48
*WM: Sravya
*WM: -78
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Parameter
*WM: hostname
*WM: 
*WM: Parameter
*WM: passcode
*WM: 
*WM: Parameter
*WM: port_str
*WM: 80
*WM: Parameter
*WM: user_id
*WM: 
*WM: static ip
*WM: 10.0.1.10
*WM: static gateway
*WM: 10.0.1.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.10
*WM: Connection result: 
*WM: 3
Should save config
WiFi connected! User chose hostname '' passcode '' and port '80'
 config...
{"hostname":"","passcode":"","port_str":"80","user_id":"","ip":"10.0.1.10","gw":"10.0.1.1","sn":"255.255.255.0"}
mDNS started. Hostname is set to .local
Local IP: 10.0.1.10
URL to send commands: http://.local:80
HTTP Server started on port 80
Turning off the LED to save power.
Error retrieving external IP
External IP address request took 8999 ms
Ready to send and receive IR signals
**it is not connecting to my router as it is showing is connected but im unable to access ip 
i have even checked devices IP Address connected to my router there is no ESP IP   **

mdhiggins, 2017-11-16:
Did you set any of the WiFi settings? It looks like you left the default IP address settings and didn’t set any hostname or port information based on those logs. The IP address needs to be compatible with your router. 

What’s your routers local IP address?

pranaystark, 2017-11-16:
Local IP: 192.168.0.1
URL to send commands: http://pranay.local:8080
HTTP Server started on port 8080
Turning off the LED to save power.
**# Error retrieving external IP
External IP address request took 8997 ms**
Ready to send and receive IR signals
[image]
# even after sending command from remote and refreshing that page it is not getting updated

mdhiggins, 2017-11-16:
So you should probably review some basic networking info or TCP/IP information before going further. It will make your future endeavors with this projects much easier. I don’t say this to be rude but I can see why you’re struggling here. 

You currently have the IP address of your device set to what I assume is the same address as your router. Assuming your router is 192.168.0.1 then your IP address for the device needs to be 192.168.0.# where # is some unused number from 2-255. The default gateway will match your router IP 192.168.0.1 and the subnet is most likely 255.255.255.0

Until the IP issues are fixed nothing else will work because it won’t be able to establish a network connection.

pranaystark, 2017-11-17:
Thankfully! it worked i have changed IP's address. But your IR Blaster code Awesome!

closed 2017-11-18 by mdhiggins