mdhiggins/ESP8266-HTTP-IR-Blaster #75: Address

groovington, 2017-11-18:
How do I send the "Address" portion of the IR code? Whenever I capture a code it has an address, when i send the address is just 0. The codes don't seem to work so I am guessing the address portion is relevant?

mdhiggins, 2017-11-18:
Address portion is usually irrelevant actually. What kind of code are you sending?

groovington, 2017-11-18:
Panasonic TV remote commands, They only work if i send them as raw

mdhiggins, 2017-11-18:
Ah Panasonic is the one exception where address does matter. You can add the address by adding &address= to your URL

groovington, 2017-11-18:
Sending http://10.10.10.249:80/msg?code=40040100BCBD:PANASONIC:48&address=0x4004&pulse=10&pass=****

Still shows the address as 0, and still doesnt work.

dham1012, 2017-11-18:
I've found that you need to convert the address from hex to dec. Try using 16388 as the address.

groovington, 2017-11-18:
{"type":PANASONIC,"data":0100BCBD,"length":48,"address":16388,"pulse":10}

its getting there but something isn't quite right. The TV is responding now, but not actually getting the right command by the looks of it.

dham1012, 2017-11-18:
try 
`{"type":PANASONIC,"data":100BCBD,"length":48,"address":16388,"pulse":10}`

dham1012, 2017-11-18:
Looks very similar to my Panasonic codes, to turn on I use
`{"type":PANASONIC,"data":100BCBD,"length":48,"address":16388,"pdelay":30,"pulse":5}`
off is 
`{"type":PANASONIC,"data":100BCBD,"length":48,"address":16388}`

mdhiggins, 2017-11-18:
You also do not need to use json to send the address, this should work
`http://10.10.10.249/msg?code=40040100BCBD:PANASONIC:48&address=16388&pulse=10&pass=xxxx`

Also I'll push down an update that will allow HEX or DEC

mdhiggins, 2017-11-18:
Update pushed

mdhiggins, 2017-11-18:
[image]

Looks like the address is displaying as it should, don't have a Panasonic TV to test on however

mdhiggins, 2017-11-21:
Any progress?

dham1012, 2017-11-21:
I've not chance to update my live unit to test sending a code and it displaying hex or dec. Will try and test later this evening and let you know.

dham1012, 2017-11-21:
Can confirm that using HEX or DEC for address works as expected and transmits and is understood by (in my case Panasonic) TV. DEC is always displayed in web interface regardless if HEX is used.

mdhiggins, 2017-11-21:
I pushed an update this morning that makes it always display HEX so it’s consistent across the board but glad to hear it’s working

closed 2017-11-28 by mdhiggins