mdhiggins/ESP8266-HTTP-IR-Blaster #79: Using on a Wemos D1 Mini - wifi config wont start

sjp770, 2017-11-20:
This is what I get when I try to start into config mode straight after flashing:

```
ESP8266 IR Controller
Config pin GPIO16 set to: 0
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
mounted file system
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.10
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
Entered config mode
0.0.0.0
IR Controller Configuration
*WM: 
*WM: Configuring access point... 
*WM: IR Controller Configuration
*WM: AP IP address: 
*WM: 192.168.244.1
*WM: HTTP server started
```
Ive change the pins around a bit because there are less on the D1 Mini, and i soldered up a little pcb that way. The Config button isnt soldered to anything - cant get it to hand out wifi. Worked great on my larger NodeMCU Amica

mdhiggins, 2017-11-20:
Looks like its in access point mode and hosting WiFi as it should be
For whatever reason instead of hosting on 192.168.4.1 its hosting on 192.168.244.1 maybe that’s the issue? But its definitely in AP mode

From: sjp770
Sent: Monday, November 20, 2017 7:31 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Subscribed
Subject: [mdhiggins/ESP8266-HTTP-IR-Blaster] Using on a Wemos D1 Mini - wificonfig wont start (#79)

This is what I get when I try to start into config mode straight after flashing:
ESP8266 IR Controller
Config pin GPIO16 set to: 0
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
mounted file system
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.10
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
Entered config mode
0.0.0.0
IR Controller Configuration
*WM: 
*WM: Configuring access point... 
*WM: IR Controller Configuration
*WM: AP IP address: 
*WM: 192.168.244.1
*WM: HTTP server started
Ive change the pins around a bit because there are less on the D1 Mini, and i soldered up a little pcb that way. The Config button isnt soldered to anything - cant get it to hand out wifi. Worked great on my larger NodeMCU Amica

sjp770, 2017-11-20:
But an ssid scan doesn't show anything. Due to the reduction in pins can I
remove the extra transmit pins?

mdhiggins, 2017-11-21:
You can change the pins around to whatever you want and you can delete additional transmission pins as needed

mdhiggins, 2017-11-28:
Any updates? Gonna close this due to inactivity, feel free to reopen

closed 2017-11-28 by mdhiggins

sjp770, 2017-11-28:
Not solved. Wont enter config mode, when the config  pin is pulled low. Can I hard code the initial ssid and password?