mdhiggins/ESP8266-HTTP-IR-Blaster #80: German Alexa App

Stef293, 2017-11-20:
Hi,

i found this great work and buyd the components for build the board, what now is ready. But i have read to late that you only provide the us alexa app. it's possible to create a german version?

thx Stef

mdhiggins, 2017-11-21:
Amazon just recently opened up the option to do this. The current beta version is pending certification from Amazon and will expand to UK/IN/CA

After that once the project is unlocked I will work on publishing on the German store

If you're interested and wouldn't mind, I do need someone who speaks German to translate the skill instructions/description

juliusedward, 2017-11-25:
I can help with IN and looking forward to this project. Please let me know the status for IN.

Thanks

mdhiggins, 2017-11-25:
I do need some help actually
I need to a couple things translated to German for the German skill

The follow statements would be great

```
IR Controller Smart Home Skill
```

```
IR Blaster Smart Home Skill that allows registration of your TV / other infrared smart home controller devices with Alexa. Your http URLs are translated into natural language smart home operations.

Register your IR Controller smart devices at https://tehpsyc.pythonanywhere.com/

Skill is designed to work with the https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster schematic and software. It may be possible to use similar HTTP enabled devices as well. The IR Blaster can be easily assembled with parts under $10 and provides full home theater control.
```

```
Alexa, turn on bedroom TV
Alexa, mute bedroom TV
Alexa, turn up the volume on bedroom TV by 5
```

For the Indian and German skill, I need someone to make me a test Amazon account and send me the credentials so I can set up a test device for Amazon to use to test the skill, please PM those details if you'd be so kind

Stef293, 2017-11-26:
I hope it helps...

Alexa, turn on bedroom TV
1.1. Alexa, schalte den Schlafzimmer Fernseher ein
1.2. Alexa, schalte den Schlafzimmer TV ein

Alexa, mute bedroom TV
2.1.Alexa, Schlafzimmer Fernseher Ton aus
2.2. Alexa, schalte den Schlafzimmer Ton aus

Alexa, turn up the volume on bedroom TV by 5
3.1. Alexa, Schlafzimmer Fernseher Lautstärke 5
3.2. Alexa, Schlafzimmer TV Lautstärke 5

Stef293, 2017-11-26:
alternative to point 2
Alexa, mute bedroom TV
2.3.Alexa, Schlafzimmer Fernseher lautlos

mdhiggins, 2017-11-27:
Submitted to Amazon, keep fingers crossed for approval

mdhiggins, 2017-12-02:
Approved

closed 2017-12-02 by mdhiggins