mdhiggins/ESP8266-HTTP-IR-Blaster #83: Delay slows down website

bimbo385, 2017-11-26:
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/blob/9b5f617836af4d7a89bdf074e623e0d7e996970e/IRController/IRController.ino#L1405

What is the reason for this delay? The website needs really long to load. I reduced this delay to 10ms, that approves the website speed a lot.

bimbo385, 2017-11-26:
improves of course...

mdhiggins, 2017-11-27:
Its just there to reduce power consumption, should only introduce at maximum a 200ms delay before the website responds, I'll do some testing to see if its interfering with the sendContent each loop though

That being said it can be safely removed if you don't want it there

mdhiggins, 2017-11-27:
I see no difference in website response time with or without the delay, it at most will create 200ms as expected, feel free to remove it if needed

closed 2017-11-27 by mdhiggins