mdhiggins/ESP8266-HTTP-IR-Blaster #87: Compile error

Twist3d, 2017-12-02:
Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2, Disabled, None, 115200"

C:\Program Files (x86)\Arduino\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino: In function 'bool setupWifi(bool)':

IRController:364: error: use of deleted function 'ESP8266WebServer& ESP8266WebServer::operator=(const ESP8266WebServer&)'

     server = ESP8266WebServer(port);

            ^

In file included from C:\Users\x\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:17:0,

                 from C:\Program Files (x86)\Arduino\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:8:

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:69:7: note: 'ESP8266WebServer& ESP8266WebServer::operator=(const ESP8266WebServer&)' is implicitly deleted because the default definition would be ill-formed:

 class ESP8266WebServer

       ^

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:69:7: error: use of deleted function 'std::unique_ptr<_Tp, _Dp>& std::unique_ptr<_Tp, _Dp>::operator=(const std::unique_ptr<_Tp, _Dp>&) [with _Tp = HTTPUpload; _Dp = std::default_delete<HTTPUpload>]'

In file included from c:\users\x\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:81:0,

                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/FS.h:24,

                 from C:\Program Files (x86)\Arduino\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:1:

c:\users\x\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\unique_ptr.h:274:19: error: declared here

       unique_ptr& operator=(const unique_ptr&) = delete;

                   ^

exit status 1
use of deleted function 'ESP8266WebServer& ESP8266WebServer::operator=(const ESP8266WebServer&)'

mdhiggins, 2017-12-02:
This looks like a problem with your Arduino installation or one of the installed libraries. I would reinstall and try again

Twist3d, 2017-12-02:
I uninstalled and deleted the libraries.  I downloaded the portable client and reinstalled the libraries/core board and I still get the same error.

mdhiggins, 2017-12-02:
The error is coming from the ESP8266WebServer class not from the script itself. That class is part of the arduino core perhaps try reinstall that?

Twist3d, 2017-12-02:
Yeah, I deleted everything again and reinstalled the core and it worked. Not sure what happened the first two times.

mdhiggins, 2017-12-02:
Glad it’s working

closed 2017-12-02 by mdhiggins