mdhiggins/ESP8266-HTTP-IR-Blaster #90: 'WiFiEventStationModeDisconnected' does not name a type

lampenschakelen, 2017-12-05:
Hi Mdhiggins, 

Thanks a lot for making this program. 

I get the follow error. 

'WiFiEventStationModeDisconnected' does not name a type. 

I used Arduino 1.8.0 and also  Arduino 1.8.5. 
I din't find answer to fix this problem.

mdhiggins, 2017-12-05:
Do you have the latest version of WiFiManager installed?

mdhiggins, 2017-12-05:
Actually make sure you’re using the ESP8266 Arduino Core 2.4.0. That might fix it

mdhiggins, 2017-12-06:
Seems to work on 2.3.0 actually

lampenschakelen, 2017-12-06:
Hi Thanks a lot for you fast response. 

WifiManager by Tzapu versie 0.12.0 Installed. 

**I fix the problem with the following instructions below.** 

Installing with Boards Manager
Starting with 1.6.4, Arduino allows installation of third-party platform packages using Boards Manager. We have packages available for Windows, Mac OS, and Linux (32 and 64 bit).
Install the current upstream Arduino IDE at the 1.8 level or later. The current version is at the Arduino website.
Start Arduino and open Preferences window.
Enter http://arduino.esp8266.com/stable/package_esp8266com_index.json into Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.
Open Boards Manager from Tools > Board menu and install esp8266 platform (and don't forget to select your ESP8266 board from Tools > Board menu after installation).

After that i search for esp8266 
Find the esp8266 by esp8266 community click update.

closed 2017-12-06 by mdhiggins