mdhiggins/ESP8266-HTTP-IR-Blaster #91: Was not declared in this scope.

lampenschakelen, 2017-12-06:
Now Following problem is coming up. 

I hope you can help me out.  

**I double checked the follow items for installation.**
Install the following libraries from the Arduino IDE Library Manager: ESP8266WebServer ESP8266WiFi ArduinoJson WiFiManager NTPClient IRremoteESP8266 as well as Cryptosuite which is not on the IDE

**All the libraries are the latest version.**

Was not declared in this scope: 
case SANYO_LC7461: output = "SANYO_LC7461";       break;  
case DENON:        output = "DENON";              break;
case COOLIX:       output = "COOLIX";             break;

Class decode_results Has no member named 'repeat'
 if (results->repeat) Serial.print(" (Repeat)");

Thanks a lot for support. 

Lampenschakelen

mdhiggins, 2017-12-06:
This looks to be a problem with your IRRemote Library.  I would reinstall or verify your installation or that. I see changes regarding the repeat parameter.

lampenschakelen, 2017-12-06:
**This fix the problem with the IR CODES**
I did a new installation of Library IRRemote. 

**New problem is coming up**

Arduino: 1.8.5 (Windows), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\**\Documents\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino: In function 'void cvrtCode(Code&, decode_results*)':

IRController:1080: error: 'RAWTICK' was not declared in this scope

       r += results->rawbuf[i] * RAWTICK;

                                 ^

C:\Users\**\Documents\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino: In function 'void dumpRaw(decode_results*)':

IRController:1129: error: 'RAWTICK' was not declared in this scope

     uint32_t x = results->rawbuf[i] * RAWTICK;

                                       ^

C:\Users\**\Documents\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino: In function 'void dumpCode(decode_results*)':

IRController:1162: error: 'RAWTICK' was not declared in this scope

     Serial.print(results->rawbuf[i] * RAWTICK, DEC);

                                       ^

Multiple libraries were found for "IRremoteESP8266.h"
 Used: C:\Users\**\Documents\Arduino\libraries\IRremoteESP8266
 Not used: C:\Users\**\Documents\Arduino\libraries\arduino_244590
exit status 1
'RAWTICK' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mdhiggins, 2017-12-06:
How are you installing IRRemote? This is another error coming from the IRRemote library that doesn’t make sense

lampenschakelen, 2017-12-07:
**Removed IRremoteESP8266** 
**Did fresh new installation** 
Include Library -->  Manage libraries --> search for IRremoteESP8266 
Now it's done compiling.

Finally it's working thanks so much for your good support.

lampenschakelen, 2017-12-07:
**Receiving IR signals is working perfectly.** 
Sending  ir signals is not working. 

When i try change the IR transmitter for a normal led i see the led flashing.  
But when i change it back for the IR led it's not received by the  wemos d2 mini with a IR receiver. 
When i press remote control the led is flashing that i'ts receiving. 

**I double check the resistors and all the cables every thing is connect to number D2.** 

I hope you can help me out

lampenschakelen, 2017-12-07:
[image]

is used the picture above.

mdhiggins, 2017-12-07:
Check and see if the LED works with a camera, most phone cameras can detect IR (iPhone front camera works, rear camera does not, has an IR filter)

Also send a picture of your actual build and let me see if its assembled correctly; possible that you reverse the poles on your IR LED?

mdhiggins, 2017-12-07:
Also are you transmitting and receiving to the same device or 2 different devices? Its designed to ignore its own signal

lampenschakelen, 2017-12-07:
I test it 2 different devices. 
Also with front camera  i see very small light flash..
I think the IR transmitter gets to less power.  I used everything in you schematic

mdhiggins, 2017-12-08:
Send a picture of t he build let me verify
It should get plenty of power. What resistor are you using?

lampenschakelen, 2017-12-08:
[image]
[image]
[image]
[image]

lampenschakelen, 2017-12-08:
Signal of the esp8266 IR transmitter is weaker that that of a remote control.  When i do the test with a front camera. When i do the testing with a wemos d2 to receive the IR singals It's not receiving. When i press the button of my remote control it's receiving the IR singals without any problems.  I think that's why my amplifier is not responding because the trasmitter IR singal is not strong enough.

mdhiggins, 2017-12-08:
So it looks to me like you might have your resistors switched though its very difficult to tell since I'm not sure which pictures correlate with which measurement
The low end 100 ohm (or even less, I use 10 ohm) resistor needs to be in series with the LED and the higher resistance one (I use 1000 ohms) should be connecting the data pin D2 and the middle pin of the 2N2222 transistor

If you switch those it will severely dampen the brightness of your transmitter

Those resistor values may vary from LED to LED though; I use an ultrabright IR LED and my range is pretty good, about 20 ft

mdhiggins, 2017-12-08:
Also curious what signal you're trying to send that's not working
Is it a normal encoded signal or a RAW or complicated signal like DENON etc

ermaya, 2017-12-08:
100 ohm resistor is not conected to led

mdhiggins, 2017-12-08:
Well it should be and that would certainly explain the brightness issue

mdhiggins, 2017-12-09:
I do not use node red, no
Hadn't really heard of it to be honest
But glad you got everything working

closed 2017-12-09 by mdhiggins

lampenschakelen, 2017-12-09:
Thanks it's working perfect with NodeRed ;-) 

[image]